dukus/digiCamControl #89: Mouse scroll wheel to cycle through images

rioforce, 2015-02-02:
I do a lot of stop-motion with DigiCamControl, and I was wondering it it would be possible to use the mouse scrollwheel to cycle between previous captured images. This would help not only stop-motion animators, but also other photographers so they can easily compare differences between photos.

Also, when using the scroll wheel, it should keep the capture button (or last clicked button) active, so I can just press spacebar to take another frame without having to go back up and click the button again.

Do you understand? Thanks, your software is great. :)

dukus, 2015-02-04:
Hi, try this version https://dl.dropboxusercontent.com/u/40987577/NCC/digiCamControlsetup_1.2.60.0.msi if you move mouse in top bar you can use mouse scroll to advance between captures. On image the mouse scroll will zoom the image. In Settings->Triggers  you can configure a key to trigger the capture .
A other option can be to use the Tools->Image Sequencer

rioforce, 2015-02-07:
It works well, so far. I haven't plugged in my camera to it yet, but from the images I already had in the folder, it worked nicely. Thanks. :)

closed 2015-02-08 by dukus