dukus/digiCamControl #114: Documentation

PrestonAshton, 2015-07-28:
Hi! I am looking to see if there is any documentation for this library published by you on Nuget. I want to make my own app for my own needs and I was wondering if there are any basic guides for camera detection. Once I have that I think I know what to do, but any further help also would be appreciated.

Thank You

-Preston

dukus, 2015-07-28:
There is a example project : https://github.com/dukus/digiCamControl/tree/master/CameraControl.Devices.Example 
In the near  future i will try to make a simple documentation.

PrestonAshton, 2015-07-28:
Thank you! :smile:

PrestonAshton, 2015-07-28:
I've been doing some work on stuff and I'm now getting an error:

An exception of type 'System.IO.FileNotFoundException' occurred in PortableDeviceLib.dll but was not handled in user code

`Additional information: Could not load file or assembly 'Interop.PortableDeviceApiLib, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.`

Do you know what this is caused by and how I can fix it? Am I missing some SDK files or something?

My camera is a Nikon D3200.

Thanks for any support and support thus far.

dukus, 2015-07-28:
Seems it is some problems with nuget  package, i will try to fix today or tomorrow

PrestonAshton, 2015-07-28:
Ah OK, thank you. I guess I should compile it myself?

dukus, 2015-07-28:
Yes then will work

PrestonAshton, 2015-07-28:
Thanks again for the quick response :sweat_smile:

dukus, 2015-07-28:
If you have any question or suggestion just let me know

PrestonAshton, 2015-07-28:
Just in time actually, my captureComplete function isn't working.

I'm adding a callback here
            currentCamera.PhotoCaptured += captureComplete;

but it never gets called once I take the shot with

            currentCamera.CapturePhotoNoAf();

Thanks for all your help so far, I really appreciate it.

dukus, 2015-07-28:
Try to use:

DeviceManager.PhotoCaptured += DeviceManager_PhotoCaptured;

PrestonAshton, 2015-07-28:
It's still not working

My code is here:

https://github.com/PrestonAshton/bCameraControl

dukus, 2015-07-28:
The DeviceManager.PhotoCaptured += captureComplete; should be called once in Form1() constructor 

The D3200 cant record directly to pc so you should set currentCamera.CaptureInSdRam = false; in connectCamera()

Just now i can't try you code but tomorrow i will make time for it

PrestonAshton, 2015-07-28:
I set that and I have done that, it still makes no difference. It seems like the first line of the callback gets triggered;
http://puu.sh/jguNw/91dbab01c7.png

but nothing ever gets added to my listview and none of the code following it is triggered.

Thanks for your help so far. :smile:

PrestonAshton, 2015-07-29:
Looking through the error log:

    Event queue processing error    at System.Windows.Forms.Control.get_Handle()
       at System.Windows.Forms.Control.OnEnabledChanged(EventArgs e)
       at System.Windows.Forms.ButtonBase.OnEnabledChanged(EventArgs e)
       at System.Windows.Forms.Control.set_Enabled(Boolean value)
       at bCameraControl.Form1.captureComplete(Object sender, PhotoCapturedEventArgs e) in     J:\foo\bCameraControl\bCameraControl\Form1.cs:line 102
       at CameraControl.Devices.CameraDeviceManager.cameraDevice_PhotoCaptured(Object sender, PhotoCapturedEventArgs eventArgs) in e:\devel\vs2010\CameraControl\CameraControl.Devices\CameraDeviceManager.cs:line 302
   at CameraControl.Devices.BaseCameraDevice.OnPhotoCapture(Object sender,     PhotoCapturedEventArgs args) in     e:\devel\vs2010\CameraControl\CameraControl.Devices\BaseCameraDevice.cs:line 442
       at CameraControl.Devices.Nikon.NikonBase.GetEvent(Object state) in     e:\devel\vs2010\CameraControl\CameraControl.Devices\Nikon\NikonBase.cs:line 1763

dukus, 2015-07-29:
The library run a event listener in a separated thread so event generated by it also run in a separated thread so if you want to interact with winform controls you should use :  BeginInvoke

For your case:
```C#
        public void captureComplete(object sender, PhotoCapturedEventArgs e)
        {
            MethodInvoker method = delegate
            {
                if (autoBracketingEnabled.Checked)
                {
                    if (nextBracket != brackets.Length)
                    {
                        nextBracket++;
                        bracket(brackets[nextBracket]);
                    }
                    else
                    {
                        captureButton.Enabled = true;
                    }
                }
                else
                {
                    captureButton.Enabled = true;
                }
            };
            if (InvokeRequired)
                BeginInvoke(method);
            else
                method.Invoke();
        }
```

PrestonAshton, 2015-07-29:
Thanks for the reply, you can tell I'm not a C# developer.
It's working now! :smile: 

Thanks for your continued support.

PrestonAshton, 2015-07-29:
I do not think I am setting the exposure compensation correctly however as it is not working. Is there a proper way to do this?

dukus, 2015-07-29:
The currentCamera.ExposureCompensation.SetValue() should use string values from 
currentCamera.ExposureCompensation.Values list. So you can use something like :
currentCamera.ExposureCompensation.SetValue("+1")

PrestonAshton, 2015-07-29:
Thank you! It's working! :smile:

PrestonAshton, 2015-07-29:
Is there way to make it take the shots quicker, as in not wait for it to save on the SD but wait for the buffer?

dukus, 2015-07-29:
You can try to call capture in row without using events, but some cases may fail if the device is busy (the writing to card isn't cause device busy)

            currentCamera.ExposureCompensation.SetValue("+1");
            currentCamera.CapturePhotoNoAf();
            currentCamera.ExposureCompensation.SetValue("+2");
            currentCamera.CapturePhotoNoAf();

PrestonAshton, 2015-07-29:
Yeah I'm getting an 'Access Denied'

PrestonAshton, 2015-07-29:
Doesn't even want to take 3 photos on a D7000, it just takes one...

PrestonAshton, 2015-07-29:
I can set the camera on 'Burst Mode' through PTP with 

currentCamera.AdvancedProperties[2].SetValue("3.0");

unfortunately I cannot find a way to change exposure through this. Although it does take the pictures fast :p:

dukus, 2015-07-29:
You can't set any properties during burst capture

PrestonAshton, 2015-07-29:
I can set the camera to continuous shooting mode and it works fine!
                            currentCamera.AdvancedProperties[3].SetValue(currentCamera.AdvancedProperties[3].Values[1]);
    currentCamera.ExposureCompensation.SetValue(exposures[i]);
    currentCamera.CapturePhotoNoAf();

This works for me and has made it so that braceting works amazingly. Is there a better way to do this?

PrestonAshton, 2015-07-29:
Its not working randomly out of the blue now... Ugh...

PrestonAshton, 2015-07-29:
Do you have any ideas why it sometimes doesn't work and sometimes it does? It more often than not doesn't... :confused:

PrestonAshton, 2015-07-30:
I have fixed it by doing:

            currentCamera.CaptureInSdRam = false;
            for (int i = 0; i < exposures.Length; i++)
            {
                try
                {
                    captureButton.Enabled = false;
                    currentCamera.AdvancedProperties[3].SetValue(2);
                    currentCamera.ExposureCompensation.SetValue(exposures[i]);
                    currentCamera.CapturePhotoNoAf();
                    currentCamera.CaptureInSdRam = true;
                }
                catch
                {
                    i = i - 1;
                }
            }
            currentCamera.ExposureCompensation.SetValue("0.0");
            currentCamera.CaptureInSdRam = false;

PrestonAshton, 2015-07-30:
What does CaptureInSdRam do?

dukus, 2015-07-30:
Will capture the image in camera Ram not in card if is true, but you take care to download the image from camera because the buffer will fill up.
https://github.com/dukus/digiCamControl/blob/master/CameraControl.Devices.Example/Form1.cs#L97

PrestonAshton, 2015-07-30:
Hmm, it just makes my camera freeze... Did you say the D3200 doesn't support it?

dukus, 2015-07-30:
Yes the D3200 not support it

PrestonAshton, 2015-07-30:
Before I give it a go, would it work on a D7000?

dukus, 2015-07-30:
Yes

PrestonAshton, 2015-07-30:
Right, I'll go and give it a go.

PrestonAshton, 2015-07-30:
All that happens is it flashes the auto focus light when it tries to take a photo...

closed 2015-08-28 by dukus