dukus/digiCamControl #125: Memory leak in image preview

francisrath, 2015-08-11:
I have problems with the application crashing after running time lapse for some time (Nikon D3200, fine JPEG, download automatically). I am running on 5 sec intervals, and after some hours the application crashes because it is out of memory.

I have found that disabling the automatic preview pane on the bottom of the screen makes the application stable.

Building with VS2015 Enterprise and running a53541f6b63365759b49cc3a3a78825206a88d6a at the moment.

dukus, 2015-08-15:
Can you check again with latest source ?

francisrath, 2015-08-16:
After a real quick test and looking at the memory profile in VS2015 it seems to be solved. Will get back when the test has run for a few hours and I will be 100% sure. Thanks for the quick fix :)

closed 2015-08-28 by dukus