dukus/digiCamControl #142: Nikon D600 - Wrong exposure delay

MasterPIC, 2015-10-22:
Hi,
I see wrong exposure delays on my Nikon D600:
[digiCamControl delay <-> effective delay (on Nikon D600)]
3s <-> OFF
2s <-> 1s
1s <-> 2s
OFF <-> 3s

Thanks
Armando Beneduce

MasterPIC, 2015-11-11:
Maybe the attached code is a good workaround...
[NikonD4.cs.txt](https://github.com/dukus/digiCamControl/files/31456/NikonD4.cs.txt)
[NikonBase.cs.txt](https://github.com/dukus/digiCamControl/files/31457/NikonBase.cs.txt)
[NikonD800.cs.txt](https://github.com/dukus/digiCamControl/files/31458/NikonD800.cs.txt)

dukus, 2015-11-18:
Please check if is working

MasterPIC, 2015-11-25:
For my own code tests I simply overwrite the involved files (CameraControl.exe, CameraControl.Core.dll and CameraControl.Devices.dll) and I'm able to execute digiCamControl.
But I've some troubles trying to execute your (updated) code without the setup.exe: if I compile the updated solution and then overwrite the whole CameraControl folder, then digiCamControl crashes. Maybe some dll needs to be registered.

MasterPIC, 2015-11-30:
I just tried 1.3.52 beta release. I found only ON and OFF values in place of OFF, 1s, 2s, 3s.

closed 2015-12-05 by MasterPIC