dukus/digiCamControl #147: Question about the development of DigiCamControl

FloBar, 2015-11-24:
Hi Duka,

It seems that the same instance Interop.WIA is defined twice under the Project References of CameraControl. This leads to a compilation error. Unfortunately, I do not know how to fix that.

Additionally, I would like to make the following changes to the program:
-	Access further settings regarding the metering mode. More precisely, I would like to add a contextual menu to select the diameter of the center-weighted area (6, 8, 10 or 13 mm), when using a Nikon D7100 in center-weighted metering mode.
-	Give the user the option to select the path where the time lapse pictures are saved.
-	Save automatically time lapse pictures with the current date and time in the format YYYY_MM_DD_hh-mm-ss.jpg (year, month, day, hour, minute, second) inside an automatically created folder named YYYY_MM_DD, which would be located in a location defined by the user.
-	Ideally write a history of the time lapse images saved within a log file with the name of the image and the time when the image was saved in this format: YYYY_MM_DD_hh-mm-fff (year, month, day, hour, minute, second, millisecond) to check the time elapsed between two shots.
-       Make sure there is 1 second +/- 100 millisecond in average between two consecutive time lapse shots, when shooting a 1 second time lapse. If that is not the case, amend the program to use a precise clock and keep the time lapse in range.

I do not necessarily expect you to make the above mentioned modifications for me but I would greatly appreciate if you could indicate me where I should be looking at in the program to make all of these modifications.

Thank you so much.

dukus, 2015-11-26:
For me working fine, send me the compilation you get

1. I just added this property as Advanced properties
2. Any time the save path can be changed using Edit Session

FloBar, 2015-11-27:
Hi Dukus,

Thank you very much. I am using VS Community 2013 Update 5. When I try to compile the latest version of the code, I cannot generate any .exe file at all and I get the following errors in my Error List:

Error	5	Metadata file 'C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\CameraControlCmd\bin\Debug\CameraControlCmd.exe' could not be found	C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\CameraControl\CSC	CameraControl

Error	6	The command "C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\PostBuild.Bat C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\ Debug" exited with code 1.	CameraControl

Error	7	The type or namespace name 'Ionic' could not be found (are you missing a using directive or an assembly reference?)	C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\CameraControl.Plugins\ExportPlugins\ExportToZip.cs	43	7	CameraControl.Plugins

Error	10	Metadata file 'C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\CameraControl\bin\Debug\Plugins\CameraControl.Plugins\CameraControl.Plugins.dll' could not be found	C:\Users\261362H\Desktop\Codes\digiCamControl-master\digiCamControl-master\CameraControlCmd\CSC	CameraControlCmd

1. Thank you for adding the new feature. I will test it with my Nikon as soon as I can compile the complete solution or run an updated CameraControl.exe.
2. Thank you for this precision regarding the save path. I still have a couple of queries though. Sorry for being so fussy:
 - Actually, I intend to use the time lapse feature to save pictures taken every second for several months, for the purpose of my research project. Thus saving all of them within the same folder would be unmanageable, due to their large number. Browsing through them would easily make Windows Explorer crash. Could you please be so kind to add in "Edit Current Session..." the option to create automatically sub-folders named [Date yyyy-MM-dd] within the user-defined save path. This way, all the pictures of the same day could be saved in the same sub-folder, in the right place.
- Also, it would be extremely convenient to have the option [Date yyyy-MM-dd-hh-mm-ss] among the formats for the file name templates regarding the saved pictures.

Thank you very much.

dukus, 2015-11-27:
1. The compilation should works now.
2. You can define template for folder too, something like: `[Date yyyy-MM-dd]/[Date yyyy-MM-dd-hh-mm-ss]` i don't know if the application will handle such quantity of files in a single session but anyway you should turn of the auto preview and/or thumbnail loading 
3. Added

FloBar, 2015-11-28:
Thank you so much, Dukus.
1. Thanks a lot for the new source code. I could successfully test the new center-weighted area dimension with the CameraControl.exe file from your last compilation. However, I still get one single last build error that prevents me from compiling:

Error	30	The command "C:\Users\261362H\Downloads\digiCamControl-master\digiCamControl-master\PostBuild.Bat C:\Users\261362H\Downloads\digiCamControl-master\digiCamControl-master\ Debug" exited with code 1.	CameraControl

Sorry for still bothering you with that but what would advise me to do to compile?

2. Indeed. This is working and this is great. You are right, I do not need the auto preview or thumbnail loading. Shall I disable them through File->Settings->Preview "Auto preview" unticked and "Don't load thumbnails" ticked?
3. Thank you so much. This is fantastic!

4. Could you please write into the log file app.log or in another log file the exact times (featuring the date and time with millisecond precision) when each time lapse picture is shot?
It seems to me there is quite some irregularities regarding the time elapsed between two consecutive shots when the time between captures is set to 1s. It looks like the camera is sometimes too fast (less than a second) in 8% of cases and too slow (more than a second) in 6% of cases. It would be thus interesting to measure the exact regularity of the time lapse with the log file.

dukus, 2015-11-28:
Still not perfect but now the build should works (Maybe you should give twice Rebuild Solution)
1. Yes
3. The shot time is saved in exif info, but no milliseconds info, the template use the pc time when transfer started but this time not exact same time as the capture time and the time difference isn't constant so which time do you want in log file ? Or there a another time when the capture command was started but it is same situation like the template time

FloBar, 2015-11-28:
Perfect. You are right, the build works and the app compiles after building the solution twice. Thank you so much for this.
1. Thank you very much. I will test the app in real conditions (with a few 100,000s shots taken within one session) in the future and will let you know if something goes wrong. With 100 shots in a row so far, I didn't encounter any problem.
2. Sorry I couldn't find exif info. How can I access it? Ideally, it would be perfect to record the capture times of each shots in milliseconds. If too complicated or not possible, I would just be happy with the times when each image started to be transferred/saved as long as the time is given in milliseconds. Thank you so much and sorry for all this hassle.

dukus, 2015-11-28:
I add a option to export file list to csv in Menu->Plugins->Export->Export file list (will export the file name, transfer time, and time difference)
I can't get a reliable result with time lapse and 1 second interval

FloBar, 2015-11-29:
Thank you very much for that. Great work. This new feature comes very handy.
I checked the accuracy of the 1 second interval time lapse. It looks acceptable without being perfect. However, I suppose there are many factors related to the camera and to the transfer of data that may impact on the consistency of the time interval.
I would like to ask one last thing before closing this thread. If I wanted to further develop the way the time lapse operates, could you please indicate to me in what part of the code/in which files the capture command of the time lapse is implemented?
Thank you very much. I won't have any further questions for now after this one.

dukus, 2015-11-29:
The time lapse capture is handed in this file https://github.com/dukus/digiCamControl/blob/master/CameraControl/ViewModel/TimelapseViewModel.cs#L556

FloBar, 2015-11-29:
Thank you so much for everything. You have been extremely helpful.

closed 2015-11-29 by FloBar