dukus/digiCamControl #150: 2.0.0.0 release not including OBS plugin?

zhzhzoo, 2015-12-19:


dukus, 2015-12-19:
No, the plugin have a separated installer
http://digicamcontrol.com/phpbb/viewtopic.php?f=12&t=898

zhzhzoo, 2015-12-19:
Thanks so much for quick reply, now it works!
But it is said to be automatically installed (http://wiki.digicamcontrol.com/index.php/Live_view_streaming, https://obsproject.com/forum/resources/digicamcontrol.253/)

closed 2015-12-20 by zhzhzoo

MuriloRP, 2018-10-01:
I need the plugin but the digiCam website is down... U can send for me ?

CCreativeCND, 2018-11-25:
Is there a version of this plugin for a 64bit OBS? In the forum thread you posted it says it is only for 32 bit OBS.

adriansuri, 2019-06-04:
Hi
There seems to be two issues.
1)This plugin is written for OBS Classic, which is no longer supported
2) It is a 32 bit plugin and hence will not work (out the box) with a 64 bit program

Simply recompiling the source code might still not work, However since it uses tcpip to communicate it should not be that difficult, plus it means that a 64 bit plugin could communicate with Digicam which is a 32 bit program ,,,,,,