dukus/digiCamControl #155: D7000 Stuck in PC Mode when CaptureInSdRam=true and Transfer Fails

robnewton, 2015-12-28:
I am writing a custom application using your library. Great job on it, I'm finding it pretty easy to use thanks to your example forms app as well as your application code.

### Brief Problem Description
I need a way to recover automatically when the SD Ram is full because a captured image did not get transferred to the PC.

### Detailed Information
I am encountering an issue when CameraControl.Devices.TransferFile() sometimes fails to transfer the file to the PC and leaves the SDRam full. When this happens its usually caused by an abnormal end while the capture thread is active. Most frequently it occurs during debugging when I start and stop quickly. However sometimes it has occurred for no apparent reason during run time.

I need to find an automated recovery process for this condition. Currently to recover from it, I have to unplug the USB cable and plug it back in. On rare occasions, I have had to reboot the camera completely (pull the battery). Toggling the camera's off/on switch does not resolve it unfortunately.

When it's like this, the camera display reads: "PC" in the top right, and only shows the battery indicator and an SD Card icon other than that.

[image]

Is there a way to recover from this within the library?

One thing I have considered is resetting the USB connection using the library linked below but I feel like this is overkill. I hope there is a method I can call to clear this problem from your library instead.
http://www.codeproject.com/Articles/21503/Hardware-Helper-Library-for-C?msg=4467165#xx4467165xx

#### My Init
```csharp
            _dslr = new CameraDeviceManager();
            _dslr.CameraSelected += dslr_cameraSelected;
            _dslr.CameraConnected += dslr_cameraConnected;
            _dslr.PhotoCaptured += dslr_photoCapturedInThread;
            _dslr.CameraDisconnected += dslr_cameraDisconnected;

            // For experimental Canon driver support- to use canon driver the canon sdk files should be copied in application folder
            _dslr.UseExperimentalDrivers = false;
            _dslr.DisableNativeDrivers = false;

            _dslr.ConnectToCamera();
```

#### My Capture
```csharp

        private void dslr_captureInThread()
        {
            _captureRetries = 0;
            Thread thread = new Thread(dslr_capture);
            thread.Start();
        }

        private void dslr_capture()
        {
            bool retry;
            do
            {
                retry = false;
                try
                {
                    if (Settings.Default.simulateCapture)
                    {
                        Logger.log("DSLR capture simulation");
                        PhotoCapturedEventArgs args = new PhotoCapturedEventArgs();
                        args.CameraDevice = _dslr.SelectedCameraDevice;
                        dslr_photoCapturedInThread(null, args);
                        args = null;
                    }
                    else
                    {
                        //_dslr.SelectedCameraDevice.AutoFocus();
                        _dslr.SelectedCameraDevice.CapturePhoto();
                    }
                }
                catch (DeviceException exception)
                {
                    if ((exception.ErrorCode == ErrorCodes.MTP_Device_Busy || exception.ErrorCode == ErrorCodes.ERROR_BUSY) && _captureRetries < 3)
                    {
                        if (_dslr.SelectedCameraDevice.CaptureInSdRam)
                        {
                            Logger.log("There is a capture in SDRam");
                        }
                        Logger.log("DSLR Capture retries remaining: " + (3 - _captureRetries) + " Error:" + exception.Message);
                        // this may cause infinite loop
                        Thread.Sleep(100);
                        _captureRetries += 1;
                        retry = true;
                    }
                    else
                    {
                        Logger.log("DSLR Error: Max capture retries exceeded: " + exception.Message);
                    }
                }

            } while (retry);
        }
```

#### My Captured Event Handler
```csharp
        private void dslr_photoCapturedInThread(object sender, PhotoCapturedEventArgs eventArgs)
        {
            // to prevent UI freeze start the transfer process in a new thread
            Thread thread = new Thread(dslr_photoCaptured);
            thread.Start(eventArgs);
        }

        private void dslr_photoCaptured(object o)
        {
            PhotoCapturedEventArgs eventArgs = o as PhotoCapturedEventArgs;
            if (eventArgs == null)
                return;
            try
            {
                string capturePath = Path.Combine(basePhotoDirectory, currentSession.Key + "_" + currentSession.CurrentPhotoIndex + ".jpg");
                if (!Directory.Exists(Path.GetDirectoryName(capturePath)))
                    Directory.CreateDirectory(Path.GetDirectoryName(capturePath));

                if (Settings.Default.simulateCapture)
                {
                    Logger.log("DSLR photo captured simulation.");
                    string simulatedCapturePath = Path.Combine(Directory.GetParent(basePhotoDirectory).FullName, "simulatedCapture.jpg");
                    File.Copy(simulatedCapturePath, capturePath);
                }
                else
                {
                    eventArgs.CameraDevice.TransferFile(eventArgs.Handle, capturePath);
                }
                eventArgs.CameraDevice.IsBusy = false;
                currentSession.RecordCapturedPhoto(capturePath);
                dslr_previewPhoto(capturePath);
            }
            catch (Exception exception)
            {
                eventArgs.CameraDevice.IsBusy = false;
                Logger.log("DSLR Error: Can't download the photo from the camera.\n" + exception.Message);
            }
            Logger.flush();
        }
```

#### My Apps Log Output
You can see in my capture code above the retry mechanism that produces this output because capture fails when the camera is in this state.
```
12/28/2015 17:28:07:593 PM DSLR Capture retries remaining: 3 Error:Device MTP error: Device is busy
12/28/2015 17:28:08:077 PM DSLR Capture retries remaining: 2 Error:Device MTP error: Device is busy
12/28/2015 17:28:08:582 PM DSLR Capture retries remaining: 1 Error:Device MTP error: Device is busy
12/28/2015 17:28:09:051 PM DSLR Error: Max capture retries exceeded: Device MTP error: Device is busy
```

dukus, 2015-12-29:
You should use lock(){} in ``dslr_photoCaptured`` function this will prevent start transfer when a another transfer it in progress 
I tried multiple methods to reset usb connection no one works, i think the nikon cameras not support this features of the usb protocol, some users solve this issue using usb or lan relays. 

I have same problem in early stage of dcc but with proper lock and checking if the camera is busy or not before capture i don't have such issue a while

robnewton, 2015-12-30:
Great, thanks for the quick response. I'll add the lock to my captured handler; thanks for the tip.

I will also add some code to better keep track of an in progress capture. I agree that this should prevent the problems I'm having but I want to have a backup plan just in case.

So I ordered a USB relay based on your tip. I'll use it to switch the AC power supply to the camera instead of switching the USB. Switching the relay will reboot the camera too and since I ran into one occasion when removing the battery was the only recovery option, this will cover that too.

Here are a couple links to what I ordered for anyone else's reference. Thanks again for the help.

http://m.ebay.com/itm/For-Smart-Home-5V-USB-Relay-1-Channel-Programmable-Computer-Control-New-/281882732713?txnId=1469458459018

https://www.amazon.com/gp/aw/d/B00B94UOM0/ref=ya_aw_od_pi?ie=UTF8&psc=1

closed 2016-03-19 by dukus