dukus/digiCamControl #163: Unhandled exception when running example code

KenEucker, 2016-01-12:
When I try to run the example code, both from the repository and within a new project downloading the library from nuget, I get the following error:

"An unhandled exception of type 'System.IO.FileNotFoundException' occurred in PortableDeviceLib.dll

Additional information: Could not load file or assembly 'Interop.PortableDeviceApiLib, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified."

It brings up a prompt looking for PortableDevice.cs and when I cancel out of that the application crashes. I'm not sure if there is something I did not do during install. I have installed the application as well.

KenEucker, 2016-01-12:
The dlls are in the package folder, at the path that they are supposed to be in. I don't have the option to copy local, in the properties, for some reason. But I can't figure out why this issue is happening.

dukus, 2016-01-12:
If the file not found in bin folder you can copy it manually

KenEucker, 2016-01-12:
The only lib files it did not copy into the bin folder are the three interop dlls:

Interop.PortableDeviceApiLib.dll
Interop.PortableDeviceTypesLib.dll
Interop.WIA.dll

Copying those into the bin folder did allow me to continue running the application. Weird.
Thanks for the help!

KenEucker, 2016-01-12:
Additionally, if you change the "Embed Interop Types" property to false and "CopyLocal" to true on these references it resolves the issue.

closed 2016-01-12 by KenEucker