dukus/digiCamControl #178: Adding Arduino real event Plugin to DigiCamControl

dukus, 2016-02-21:
http://digicamcontrol.com/phpbb/viewtopic.php?f=5&t=1018

gabber1, 2016-05-19:
Relevant thread: http://digicamcontrol.com/phpbb/viewtopic.php?f=3&t=1304
Can we get an update/priority indication of implementation of this feature?

Thanks

dukus, 2016-05-19:
Please describe your ide how this should be implemented ? I started to work on this but i'm not sure if i'm in good way

gabber1, 2016-05-19:
That was fast!
For starters. we could add a new trigger device, similar to the existing Serial trigger but that uses 9600 baudrate and sends out a 'trigger character\n\r' to a COM port selected in dropdown menu. An arduino could listen to that. To expand this idea, there could be a checkbox to wait for an ACK character from the Arduino (only after the ACK the software continues). Combining this with the existing Stop Motion feature, we could build something like this: https://www.kickstarter.com/projects/alpinelabs/radian-2-bluetooth-time-lapse-motion-and-camera-co
It could then be expanded to support more difficult commands (like zooming commands, etc). Maybe add a tab in the liveview with a 'control/send serial commands' so you can send custom strings via Serial at a press of a button, but this customization is more advanced.

Hope you understand what I meant, happy to explain it in different words.
BTW, what does the string "{5B6842B3-E486-4A3E-A0C3-26988B6F0123}" mean?

dukus, 2016-05-19:
Ok i will try to implement a simple trigger device for arduino. The software can't be suspended because this mean to freeze all UI actions.

gabber1, 2016-05-20:
That makes sense. Can you create a global variable/flag to wait for when running the time lapse, so it only triggers after an acknowledgement char (defined under device) from trigger device? 
This would mean changing the code of the time lapse to work specifically for this application, in future a brother approach could be implemented that is compatible with more parts of the software.

dukus, 2016-05-27:
Can you check this version https://s3.amazonaws.com/download.digicamcontrol.com/digiCamControlsetup_2.0.36.0.exe there is a separated plugin (Menu->Plugins->Tools->Arduino) to handle arduino communication

serg3y, 2016-05-28:
I have core stability issues.
Program crashes just after main GUI appears.
Does not matter if camera is connected or not.
Initially Installed 2.0.36 atop of 2.0.27, but also tried to uninstall, delete C:\ProgramData\digiCamControl, and install 2.0.36
Log file simply stops at "RefreshDisplay":
2016-05-28 13:48:58,069 [1]DEBUG DCC [(null)] - --------------------------------===========================Application starting===========================--------------------------------
2016-05-28 13:48:58,086 [1]DEBUG DCC [(null)] - Application version : 2.0.36.0
2016-05-28 13:48:58,087 [1]DEBUG DCC [(null)] - Init : Analytics
2016-05-28 13:48:58,105 [1]DEBUG DCC [(null)] - Init : DeviceManager
2016-05-28 13:48:58,106 [1]DEBUG DCC [(null)] - Init : ExternalDeviceManager
2016-05-28 13:48:58,110 [1]DEBUG DCC [(null)] - Init : Trigger
2016-05-28 13:48:58,111 [1]DEBUG DCC [(null)] - Init : QueueManager
2016-05-28 13:48:58,115 [1]DEBUG DCC [(null)] - Init : ScriptManager
2016-05-28 13:48:58,119 [1]DEBUG DCC [(null)] - Init : PluginManager
2016-05-28 13:48:58,121 [1]DEBUG DCC [(null)] - Init : _pipeServer
2016-05-28 13:48:58,388 [1]DEBUG DCC [(null)] - Logo file not foundC:\Program Files (x86)\digiCamControl\Plugins\CameraControl.Plugins\logo.png
2016-05-28 13:48:58,389 [1]DEBUG DCC [(null)] - Loading plugin dll: C:\Program Files (x86)\digiCamControl\Plugins\CameraControl.Plugins\CameraControl.Plugins.dll
2016-05-28 13:48:59,154 [1]DEBUG DCC [(null)] - Loading plugin dll: C:\Program Files (x86)\digiCamControl\Plugins\Plugin.DeviceControlBox\Plugin.DeviceControlBox.dll
2016-05-28 13:49:03,573 [1]DEBUG DCC [(null)] - Connection device start
2016-05-28 13:49:03,725 [1]DEBUG DCC [(null)] - Window command received :RefreshDisplay
2016-05-28 13:49:03,761 [1]DEBUG DCC [(null)] - Window command received :Zoom_Image_Fit
2016-05-28 13:49:03,767 [1]DEBUG DCC [(null)] - Window command received :Zoom_Image_Fit
2016-05-28 13:49:03,950 [1]DEBUG DCC [(null)] - Window command received :RefreshDisplay

dukus, 2016-05-28:
Which video card do you have ? 
If you start the application with empty session folder in settings->Advanced  you can disable the hardware acceleration

serg3y, 2016-05-28:
I disabling hardware acceleration and seemed to have fixed it. Then I re-enabled it and it's still working!
I suspect I also clicked Advanced > "Clear cache", but not sure if that's what fixed it.
Graphics: AMD Radeon HD 7900 (ASUS), Driver: v12.104.0.0 (2013-03-28), OS: Win7 64 bit.

I also tested on Lenovo X220 Laptop and had some unusual behavior with not being able to display older images in main view, also all thumbnails showed the same image. Clearing Cache did not fix it. I deleted some sessions and it seems to have come good. I added a session and its still good. :)

Sorry, this probably doesn't help much.

serg3y, 2016-05-28:
I think there is a problem when I create a new session but DCC is still trying to show thumbnails from the old session folder, then I delete those files and it starts crashing. Hardware is not related. Clearing Cache does fix it. Log attached (note: the "DSC_0005.jpg" file mentioned in the log was deleted)
[app.log.txt](https://github.com/dukus/digiCamControl/files/287776/app.log.txt)

serg3y, 2016-05-28:
Hardware Acceleration Enabled with AMD Radeon HD 7900 causes a crash when DCC v2.0.36 attempts to make new thumbnails cache for new or existing photos.
SOLUTION: 
1) Empty the session by removing all files or renaming session folder to stop DCC from creating new creating new thumbnails cache.
2) Start DCC > File > Settings > Advanced > Disable Hardware Acceleration > Quit DCC for the change to take effect.
3) Move files or folder back.
4) Start DCC and it will generate working thumbnails cache.

closed 2016-09-10 by dukus