dukus/digiCamControl #183: Picture Filename

FloBar, 2016-03-09:
Hi,

I have developed a new more accurate timer for the time lapse.
However, I would like each picture to be saved with a name in the format "yyyy_MM_dd_HH_mm_ss" that contains the shot time (and not the transfer time).
I recall you mentioned that "The shot time is saved in exif info" in a previous thread.
Could you please help me to locate where in the code the shot time of each picture is saved? I would like to refer to this data in FilenameTemplate.cs.
Thank you so much.

dukus, 2016-03-09:
The exif data isn't generated when photo is downloaded because this will slow down the preview process, so not really can be used in filename template

dukus, 2016-03-09:
Why do you not use the latest source code ?

FloBar, 2016-03-09:
In fact, I need to take shots on my Nikon as consistently as possible, say every 3 seconds, to study the motion of objects. I save my photos on a PC equipped with a large HDD.
I actually do not need at all to preview these photos but I would need to know with seconds precision when they were shot.
I am happy to use the latest source code if it can help me to save my photos the way I want. Do you think it could possible at all to name and save photos according to their shot time (even without the help of exif data)?

dukus, 2016-03-09:
I can add a tool plugin to rename captured file based on exif data if this can help you in this way no special changes required on transfer part which already is to much complicated

FloBar, 2016-03-09:
That would be fantastic. Thank you so much. If you do add this tool, could you please tell me where you added it in the code?

FloBar, 2016-03-16:
I tried to use the latest code but only 2 projects loaded over all 17 of the solution.
Could you please help me to solve that problem?

FloBar, 2016-03-16:
OK. I could open the entire solution after installing .NET framework 4.6.
However, although I can compile, the app does not start. Do you think it is needed to use Visual Studio Community 2015 instead of Community 2013?

dukus, 2016-03-16:
I use Community 2013 but the Community 2015 is also installed

dukus, 2016-03-16:
Do you have Visual C++ Redistributable for Visual Studio 2015 installed ? (https://www.microsoft.com/en-us/download/details.aspx?id=48145) this also required

FloBar, 2016-03-16:
Thank you. Yes, I do. I have Visual C++ Redistributable for Visual Studio 2015 Version 14.0.23026 installed but I will try to reinstall it to see if it helps.

Also, I get the following error when I debug the app:
"An exception of type 'System.DllNotFoundException' occurred in CameraControl.Core.dll but was not handled in user code
Additional information: Unable to load DLL 'sqlite3': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
If there is a handler for this exception, the program may be safely continued."

Still, I could see that sqlite3.dll exists in "...\digiCamControl-master\CameraControl.Application"

arpinerap, 2016-04-01:
> Like you, I have Visual Studio Community 2013 and 2015 installed, Visual
> C++ Redistributable for Visual Studio 2015 also installed. My setup is as
> good as it can be. Still, I still cannot run the latest version of your
> code and still encounter the same problem I mentioned a couple of weeks ago:
> Also, have you added a tool plugin to rename captured file based on the
> exif data as you mentioned? Since I cannot run the program, I could not
> check that.
> The need for this extra feature is becoming pressing as I am starting to
> run out of time for my PhD research project.

Good morning.

Have you try to downgrade MahApps.Metro lib?

Now is a prerelease (1.3.?) , you have to uninstall it throw "Manage Nuget
Packages for Solution" (right click on solution) and install last Stable
release (1.2.4.0)
Proyects afected:
- CameraControl
- Core
- Plug Manager
- Plugins
- Server Test
- Test
- Trigger
- ControlBox

Hope you can compile....

FloBar, 2016-04-01:
Good day Duka,

Unfortunately, I already have the 1.2.4.0 version of MahApps.Metro lib. As a matter of fact, I can compile and generate an .exe but there seems to be something wrong with the code when I try to debug after compiling by pressing F5. The run gets interrupted by a break and the following message appears:

"An exception of type 'System.DllNotFoundException' occurred in CameraControl.Core.dll but was not handled in user code
Additional information: Unable to load DLL 'sqlite3': The specified module could not be found. (Exception from HRESULT: 0x8007007E)".

Sorry for insisting and thank you for your assistance.

dukus, 2016-04-01:
Copy CameraControl.Application\sqlite3.dll to debug folder this not related to MahApps

FloBar, 2016-04-01:
Thank you Duka. Indeed, that solved the problem. Now, I can debug the app. 
My last problem, however, is to find out what I can do so that time lapse photos be named according to the exact time when they got shot, albeit by using the exif info or through another method. Very honestly, I do not mean to use the preview process at all in my application. On the contrary, I intend to disable previews as completely as possible.
Thus, I originally thought it would be ideal to name directly photos according to the exif info prior to saving them. However, in the worst case, I am happy to rename them after get saved. In any case, I absolutely need their name to contain the time they were shot for the purpose of my research project. It is becoming rather urgent now.
Any clue of how I should alter the code to do that? Sorry for bringing back this topic and thank you so much for your help.

FloBar, 2016-04-08:
I has been a month now since I opened this thread and I am still completely stuck. Here is my last request ever. I will close this thread and not open another one if you help me with that:
How can I share a string between CameraControl.ViewModel.TimeLapseViewModel.cs and CameraControl.Core.Classes.FilenameTemplate.cs ?
I would need this string value to be defined by the former and to be read by the latter. Thank you in advance.

dukus, 2016-04-08:
For communication between modules you can classes you can use the 
`ServiceProvider.WindowsManager.ExecuteCommand("YourString_data");` 
And you can receive it using WindowsManager.Event 
`ServiceProvider.WindowsManager.Event += WindowsManagerEvent;`

If you have patient i will try to add file template based on exif data today

arpinerap, 2016-04-08:
Ok. Perfect.
If you make an example, i will read all the instructions of Zerene and work
on it.
I should need also an example of how to manage CombineZP method to make its
tab run.

I am thinking on make any kind of tutorial of how to manage the plugin.
Will you added to yours in web? Is there any specific way to do it or a
template?

Thanks so much.

FloBar, 2016-04-09:
Thank you very much for your hard work, Duka and Arpinerap. I appreciate your responsiveness. However, I cannot compile the latest source code because CameraControl.Plugins\CameraControl.Plugins.dll seems to be missing.
Thus, I simply used the previous source version and replaced the files you modified. I am very glad you added the option of naming files according to the exif date and time. Nonetheless, I cannot transfer images to a computer. When use the time lapse, no photos are saved and the following error is returned: "Transfer Error! Message {0} Object reference not set to an instance of an object."

dukus, 2016-04-10:
I can compile the source and the CameraControl.Plugins.dll isn't referenced anywhere so no sense of the missing error. 

Because your source isn't same with master i need more information about your error ....

FloBar, 2016-04-10:
OK. I downloaded the latest source and can compile now.
My source is exactly the same as the master and I still get the same error:
when I shoot time lapse photos with a local folder on my C: drive defined on my computer, no pictures are saved. The error message "Transfer Error! Message {0} Object reference not set to an instance of an object." is displayed instead. 
I just sent you my log file.

FloBar, 2016-04-14:
Thank you very much. The app does what I wanted it to do. Thank you very much for your help.

closed 2016-04-14 by FloBar