dukus/digiCamControl #213: How to use template filename in CameraControlCmd

ataheri, 2016-07-10:
In the CameraCommandCmd help it say we can use a template file for the file name. Which format should we save the file? and how many file names can we provide in the template file (more than one picture?)

ataheri, 2016-07-10:
I guess we can use the command like this:
CameraControlCmd.exe /capture /filenameTemplate [FNumber]_[Date yyyy-MM-dd-hh-mm-ss]_[Counter 5 digit]

ataheri, 2016-07-10:
Well, tried it and it is not working. Has problem with the space and brackets. Also dont know if we can use barcode data in the command version.

dukus, 2016-07-10:
You should use the remote utility http://digicamcontrol.com/doc/userguide/remoteutil which have almost same functionality but is much faster and stable

ataheri, 2016-07-10:
Thanks for the link. I tried the remoteutil like this: but it gives an error:
C:\Program Files (x86)\digiCamControl>CameraControlRemoteCmd.exe /
folder c:\pictures
Error :
The operation has timed out.
   at System.IO.Pipes.NamedPipeClientStream.Connect(Int32 timeout)
   at CameraControlRemoteCmd.CommandProcessor.Send(String sendStr,
me, String server, Int32 timeOut)
   at CameraControlRemoteCmd.CommandProcessor.Parse(String[] args)

serg3y, 2016-07-11:
Ataheri,
If you happen to use MatLab you may want to try my wrapper for CameraControlRemoteCmd.exe v2.0.27.
https://au.mathworks.com/matlabcentral/fileexchange/57196-dslr-camera-controller--wrapper-for-digicamcontrol-
Its still a bit fresh, watch out for bugs.

BStrauss3, 2016-07-13:
Try putting the whole parameter value in double quotes. I suspect that the parameter is getting truncated at the 1st space.

closed 2016-09-12 by dukus