dukus/digiCamControl #214: 'System.NotImplementedException' in CameraControl.Devices.dll

pbalut, 2016-07-11:
I have a problem with PhotoBooth project. Every time when I want to take a photo the method _CapturePhoto_ (exactly line _Device.ExecuteCommand(Conts.wiaCommandTakePicture)_) from namespace _CameraControl.Devices.Others.WiaCameraDevice_ generate exception: _Exception thrown: 'System.NotImplementedException' in CameraControl.Devices.dll_

I test it using Canon EOS 40D. 

Do you have any idea where is problem ?

dukus, 2016-07-11:
The main application it is working for you ?

pbalut, 2016-07-11:
Yes, the main app works great.

le-rou, 2016-08-01:
same for me. Tested the software with my EOS 500D, worked perfectly. Downloaded the git repo and I get the same thing. Camera is detected but taking a picture throws the not-implemented-exception :/ Any tips?

BStrauss3, 2016-08-01:
Make sure you (manually) copy the Canon EDSDK.dll into the directory where the compile is, bin\debug

-----Burton

Burton Strauss III, PMP
  mailto:BStrauss@acm.org
  +1 (646) 867-3364

le-rou, 2016-08-01:
tried that. Still getting that exception :/

BStrauss3, 2016-08-01:
When you run it from Visual Studio what do you see in the exceptions in the output window? I was seeing the failure to load message which went away when I copied the dll.

Note that which bin\debug to copy into is dependent on which program you are running, Camera Controls vs. CameraControlCmd vs. CameraControlRemoteCmd

-----Burton

Burton Strauss III, PMP
  mailto:BStrauss@acm.org
  +1 (646) 867-3364

le-rou, 2016-08-01:
I cloned the git repo and am trying to get the example to run (CameraControl.Devices.Example) as I'm looking for a simple way to just capture a picture. So i copied the EDSDK.dll into the bin/debug in the example dir. Still getting the exception:

Exception thrown: 'System.NotImplementedException' in CameraControl.Devices.dll
An unhandled exception of type 'System.NotImplementedException' occurred in CameraControl.Devices.dll
Additional information: The method or operation is not implemented.

I also tried the CameraControl now - it handles the exception but still I get the same result (see the excetption in the output windows, getting an error in the application). Tried copying the dll here, but it was already there.

Btw.: Thanks for your help so far!

BStrauss3, 2016-08-01:
Look further up the exception stack i.e. earlier in time. If it can't load the dll that exception gets swallowed and you hit the not implemented in the base class

-----Burton

le-rou, 2016-08-02:
Again thank you for taking the time!

I've tried around a bit. There is no word about the EDSDK.dll - not that it's loaded and not that it couldn't be loaded. I tried a DLLImport in the Form1.cs and called the init method. That worked. The dll is definitely in the bin/debug directory. The Canon.Eos.Framework.dll is built and present. That's the one implementing the DLLImports anyway. 

I'm kind of stuck here.. any more advice would be very much appreciated. I did make a pastebin of my output - this one is the unmodified version. Just ran the CameraControl.Devices.Example from VS and hit Capture (the EOS 500 was selected in the combo box).

http://pastebin.com/Gyq9UTAi

Thanks again!

dukus, 2016-08-02:
Please check the source again, now the debug info is outputted directly in form text box

le-rou, 2016-08-02:
Just to be sure I deleted my local version and cloned it new.

app.config is missing for the example project. Put in a dummy.

Started the example and it outputs:

Connection device start
Connection device \\?\usb#vid_04a9&pid_31cf#5&1e92747d&0&12#{6ac27878-a6fa-4155-ba85-f98f491d4f33}
Wia Camera Found: Canon EOS 500D
===========Camera is connected==============
Driver :WiaCameraDevice
Name :Canon EOS 500D
Manufacturer :Canon Inc.

Then I click on capture and get the NotImplemented Exception.

https://imagebin.ca/v/2qBvpkuFMjgA

dukus, 2016-08-02:
Check it now

le-rou, 2016-08-02:
It works. Thank you very much for your effort, it's very much appreciated!