dukus/digiCamControl #229: Barcode recognizer and writer

dukus, 2016-09-02:
https://zxingnet.codeplex.com/

BStrauss3, 2016-09-02:
So what's the plan? To allow extracting bar codes from images?? Or are you thinking of generating barcodes out of digiCamControl?  I guess I don't get the use case(s)...

dukus, 2016-09-02:
Extracting barcode from images and using  it as file name template

BStrauss3, 2016-09-02:
Interesting. seems like a bit of a rare edge case, but interesting
never-the-less. 

 

I would recommend making it an API-type item or a plugin so you could pass
it for analysis w/o having to update the main program.

 

 

 

-----Burton

 

XqE6dOkks5qmGI8gaJpZM4Jzcm7> .
<https://github.com/notifications/beacon/ATbhQauYnKCpMQ4WZfiTWP5jphMy4M2Gks5
qmGI8gaJpZM4Jzcm7.gif>