dukus/digiCamControl #243: Add /delay [ms] in single line capture utility to delay the capture

Meshz, 2016-11-05:
First of all thanks for your work!

I would like to know if it's possible to add a `/delay` parameter in the single line utility command. I am executing the capture through PHP and I experience a delay of ~2ms between `exec` and the capture of the picture.

I guess this is due to the time of connection to the camera (`New Camera is connected ! Driver xxx`). So if we could set a custom delay, we could be sure of the exact time when the picture is captured.

BStrauss3, 2016-11-05:
It would help to specify which cmd line program (there are two) - one is a pure cmd line and one just sends commands to the GUIDE backend.

Also, it would help if you can include a capture of the text you are seeing & indicate where you want the delay.

On Nov 5, 2016, Meshz <notifications@github.com> wrote:
> First of all thanks for your work!
>
> I would like to know if it's possible to add a `/delay` parameter in
> the single line utility command. I am executing the capture through PHP
> and I experience a [...]

-----Burton

Burton Strauss III, PMP
  mailto:BStrauss@acm.org
  +1 (646) 867-3364

Meshz, 2016-11-05:
@BStrauss3 I'm talking about CameraControlCmd.exe (http://digicamcontrol.com/doc/userguide/cmd). Please see the attached picture for current output of cmd.

The point of delaying is to be sure that when I execute cmd, the picture will be taken exactly `[delay]ms` after. So, it could be great to have:

_C:\>"C:\Program Files (x86)\digiCamControl\CameraControlCmd.exe" /filename "C:\test.jpg" /capture /delay 5000
digiCamControl command line utility
**Waiting 5000ms before capture**
New Camera is connected ! Driver :D3300
**Delay ended, capturing picture**
Photo transfer begin.
Transfer started :C:\test.jpg
Transfer done :C:\test.jpg
Photo transfer done._

[image]

BStrauss3, 2016-11-05:
I would have to look at the code to be a hundred percent sure but I don't think that's feasible. 

There's a delay (and its not an easy one to measure) between when you press Enter on the command and when the command line is being processed and we could figure out 5000 milliseconds from when. There's another unpredictable delay between when the camera is recognized and when the code would get control having to do with the USB drivers. I haven't looked at the Nikon code but I know with cannon its also in a separate thread.

That isn't to say can't be done but it might give you the results your looking at if instead of 5000 millisecond you gave an exact time stamp and the image was triggered on or after that point. 

/delayuntil 12:53:35.023454

If the precision is that important I would probably code it as a sleep to get close and then a spinning loop until the exact time.

-----Burton

On Nov 5, 2016, Meshz <notifications@github.com> wrote:
> @BStrauss3 I'm talking about CameraControlCmd.exe
> (http://digicamcontrol.com/doc/userguide/cmd). Please see the attached
> picture for current output of cmd.
>
> The point of delaying is to be sure that [...]

Meshz, 2016-11-06:
@BStrauss3 Thanks for your response. Using a defined timestamp is a great idea and I understand why this is more predictable than with fixed ms. The precision is an important thing as I am using it in a personal photobooth showing a countdown before photo is taken.

That said, it's not a urgent case as I am already satisfied with my implementation. I launch the capture command 2s before countdown ends, because I noticed that the capture take between 1.5 & 2.5s to be done on my local machine. Was just an idea though. 😄

dukus, 2016-11-06:
@Meshz If you use the CameraControlRemoteCmd.exe this feature almost is implemented. In main app  you can set a delay for capture in Menu=>Camera=>Camera property or using command line

`CameraControlRemoteCmd.exe /c set property.delay 1`

serg3y, 2016-11-06:
"we could be sure of the exact time when the picture is captured" so the aim is to take the pic at a specific absolute time... .
If you use CameraControlRemoteCmd.exe like Dukus said then the delay is much smaller and more repeatable ~0.035s +-0.01s for my Canon. I was able to measure this time by imaging the computer's own clock (i did not adjust for monitor delay as i don't know what it is). Then I used my MatLab wrapper to allow me to take photos on the second +-0.01s... There may be an odd photo that gets delayed by more then this if the computer decides to have a think about the meaning of life, but it seemed good for most photos.
The MatLab wrapper for CameraControlRemoteCmd.exe is here. (still beta)
https://au.mathworks.com/matlabcentral/fileexchange/57196-dslr-camera-controller--wrapper-for-digicamcontrol-

Meshz, 2016-11-06:
@dukus @serg3y In fact using `CameraControlRemoteCmd.exe` is much more faster (almost instant). I'll use this, thank you all!

closed 2016-11-08 by dukus