dukus/digiCamControl #278: Does not build using VS2017

macro63, 2017-08-30:
Latest version of code V2.0.73.5 Aug 28, 2017 does not build using VS2017
see error list below ...

Severity	Code	Description	Project	File	Line	Suppression State
Error		The tag 'CustomPropertyCollection' does not exist in XML namespace 'clr-namespace:Capture.Workflow.Core.Classes;assembly=Capture.Workflow.Core'. Line 26 Position 10.	Capture.Workflow	C:\digiCamControl-master\Capture.Workflow\Wpf\Controls\PropertyEditor.xaml	26	
Error	CS0227	Unsafe code may only appear if compiling with /unsafe	Capture.Workflow.Core	C:\digiCamControl-master\Capture.Workflow.Core\Classes\Utils.cs	50	N/A
Error		The command "cd ..\..\
"C:\digiCamControl-master\Setup\bin\Debug\Setup.exe"" exited with code -532462766.	Setup			
Error	CS0006	Metadata file 'C:\digiCamControl-master\Capture.Workflow.Core\bin\Debug\Capture.Workflow.Core.dll' could not be found	Capture.Workflow.Plugins	C:\digiCamControl-master\Capture.Workflow.Plugins\CSC	1	N/A

macro63, 2017-09-01:
I have more to add ...
I have downloaded your new zip file of source dated 31-8-2017
When I build this -> the same 4 errors.
I have solved 3 of them as follows
Right-click Capture.Workflow.Core -> Properties -> Build
Check 'Allow unsafe code'
Build  -> Clean solution -> Rebuild Solution.
This removes 3 errors, and leaves just the Setup error.

Unhandled Exception: System.Exception: WiX binaries cannot be found. Wix# is capable of automatically finding WiX tools only if WiX Toolset installed. In all other cases you need to set the environment variable WIXSHARP_WIXDIR or WixSharp.Compiler.WixLocation to the valid path to the WiX binaries.
20>  WiX binaries can be brought to the build environment by either installing WiX Toolset, downloading Wix# suite or by adding WixSharp.wix.bin NuGet package to your project.
20>     at WixSharp.Compiler.get_WixLocation()
20>     at WixSharp.Compiler.Build(Project project, String path, OutputType type)
20>     at WixSharp.Compiler.Build(Project project, OutputType type)
20>     at Setup.Script.Main(String[] args) in C:\digiCamControl-master\Setup\setup.cs:line 184
20>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(4935,5): error MSB3073: The command "cd ..\..\
20>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\Bin\Microsoft.Common.CurrentVersion.targets(4935,5): error MSB3073: "C:\digiCamControl-master\Setup\bin\Debug\Setup.exe"" exited with code 255.
========== Rebuild All: 19 succeeded, 1 failed, 0 skipped ==========

How do I solve this? This appears to me that you have additional software
installed on your pc that I and anyone else that downloads this zip file,
unpacks it, and loads the solution file into VS does not have. Cheers.

dukus, 2017-09-01:
Hi for setup creation WiX Toolset is used, 
In documentation this is specified at http://digicamcontrol.com/doc/development
You have two choices 
1.  Install WiX Toolset
2. Uncheck from build list the Setup project

macro63, 2017-09-02:
Thanks, I have downloaded and installed the WiX Toolset v3.11 and the WiX Toolset Visual Studio Extension.
The build works perfectly ...

20>------ Rebuild All started: Project: Setup, Configuration: Debug x86 ------
20>C:\digiCamControl-master\Setup\ObsPluginSetup.cs(45,42,45,44): warning CS0168: The variable 'ex' is declared but never used
20>C:\digiCamControl-master\Setup\setup.cs(135,46,135,49): warning CS0168: The variable 'exn' is declared but never used
20>  Setup -> C:\digiCamControl-master\Setup\bin\Debug\Setup.exe
20>  Searching for custom action entry points in Setup.exe
20>      Loaded dependent assembly: C:\Program Files (x86)\WiX Toolset v3.11\sdk\Microsoft.Deployment.WindowsInstaller.dll
20>      InstallCRTAction=Setup!Setup.CustomActions.InstallCRTAction
20>      MyAction=Setup!Setup.CustomActions.MyAction
20>      SetRightAction=Setup!Setup.CustomActions.SetRightAction
20>  Searching for an embedded UI class in Setup.exe
20>  Modifying SfxCA.dll stub
20>  Copying file version info from C:\digiCamControl-master\Setup\Setup.exe to C:\digiCamControl-master\Setup\%this%.CA.dll
20>  Packaging files
20>      Setup.exe
20>      CustomAction.config
20>      Setup.pdb
20>      WixSharp.dll
20>      Microsoft.Deployment.WindowsInstaller.dll
20>  MakeSfxCA finished: C:\digiCamControl-master\Setup\%this%.CA.dll
20>  Wix# support for EmptyDirectories is automatically enabled
20>  
20>  ----------------------------------------------------------
20>  
20>  Wix project file has been built: C:\digiCamControl-master\Setup\digiCamControlsetup_2.0.73.5.wxs
20>  
20>  Windows Installer XML Toolset Compiler version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  digiCamControlsetup_2.0.73.5.wxs
20>  Windows Installer XML Toolset Linker version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  
20>  ----------------------------------------------------------
20>  
20>  MSI file has been built: C:\digiCamControl-master\Setup\digiCamControlsetup_2.0.73.5.msi
20>  
20>   ProductName: digiCamControl
20>   Version    : 2.0.73.5
20>   ProductId  : {19d12628-7654-4354-a305-9ab0b32e867c}
20>   UpgradeCode: {19d12628-7654-4354-a305-9ab0932af676}
20>  
20>   Auto-generated InstallDir ID:
20>     INSTALLDIR=%ProgramFiles%\digiCamControl
20>  
20>  ----------------------------------------------------------
20>  
20>  Wix project file has been built: C:\digiCamControl-master\Setup\ObsPluginSetup_2.0.73.5.wxs
20>  
20>  Windows Installer XML Toolset Compiler version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  ObsPluginSetup_2.0.73.5.wxs
20>  Windows Installer XML Toolset Linker version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  
20>  ----------------------------------------------------------
20>  
20>  MSI file has been built: C:\digiCamControl-master\Setup\ObsPluginSetup_2.0.73.5.msi
20>  
20>   ProductName: OBS plugin for digiCamControl
20>   Version    : 2.0.73.5
20>   ProductId  : {357e0d80-5093-478e-8c11-28b1c72426ed}
20>   UpgradeCode: {357e0d80-5093-478e-8c11-28b1a72096e7}
20>  
20>   Auto-generated InstallDir ID:
[... 90 lines cut ...]
20>  Wix project file has been built: C:\digiCamControl-master\Setup\digiCamControlsetup_2.0.73.5.wxs
20>  
20>  Windows Installer XML Toolset Compiler version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  digiCamControlsetup_2.0.73.5.wxs
20>  Windows Installer XML Toolset Linker version 3.11.0.1701
20>  Copyright (c) .NET Foundation and contributors. All rights reserved.
20>  
20>  
20>  ----------------------------------------------------------
20>  
20>  Bootstrapper file has been built: C:\digiCamControl-master\Setup\digiCamControlsetup_2.0.73.5.exe
20>  
20>   Name       : digiCamControl
20>   Version    : 2.0.73.5
20>   UpgradeCode: {19d12628-7654-4354-a305-3a2057e2e6c9}
20>  
========== Rebuild All: 20 succeeded, 0 failed, 0 skipped ==========

This takes a very long time to build, so I think I will remove Setup from the build while I modify code since the compiled files are available in \Camera Control\bin\debug\

I have Canon gear, and past experience with the Canon EOS SDK (ED-SDK) and have built my own application (32bit Windows) 5 years ago.

I have some coding/functional issues with this program which I will raise separately. You certainly have some great ideas implemented here, this is way past what I ever did 5 years ago in C++

dukus, 2017-09-03:
Yes you can remove it or disabled from build in project manager

closed 2017-09-03 by dukus

dukus, 2017-09-03:
If you want and have enough free time any help is welcomed