dukus/digiCamControl #280: Camera simulation

CTugTEZ, 2017-09-14:
For learning to use digiCamControl detailed, I guess people need to a camera simulation. Do you have plans add it or it is already in but I'm missing it? By the way thank you so much for creating this amazing program. Cheers.

BStrauss3, 2017-09-14:
No camera simulation is going to give you the richness and complexity that you'll experience with a real camera. At best it would be a pale and very limited imitation.

Each camera - even similar models within the same brand - have their small differences and idiosyncrasies.

You'll get far better results hooking up the camera you're really going to use and experimenting with it. After all it's not like the old days when you would have had to buy film and pay to have it developed or developed it yourself to see the results. You can always just delete the images that are bad.

-----Burton

On Sep 14, 2017, Cihan Tugrul TEZCAN <notifications@github.com> wrote:
>For learning to use digiCamControl detailed, I guess people need to a
>camera simulation. Do you have plans add it or it is already in but I'm
>missing it? By the way thank you so much for creating this amazing
>program. Cheers.

dukus, 2017-09-16:
You can enable the "fake camera" in File->Settings->Advanced  a restart of the software is required. 
Please use this version https://s3.amazonaws.com/download.digicamcontrol.com/digiCamControlsetup_2.0.73.6.exe
But how BStrauss3 says it is almost impossible to simulate a real camera....

closed 2017-10-12 by dukus