dukus/digiCamControl #281: Saving files using only CameraControl.Devices library

YanYas, 2017-10-09:
Hello,

I'm a pretty new to c# and I'm trying to implement some features of your library in [vvvv](https://vvvv.org/propaganda). I'm able to focus and take a picture but how do I save the image to disk? 

In your Form demo you the photo isn't Saved-to-Disk and the option isn't available to choose. I can't discern a method that enables this functionality. Perhaps it went over my head.

In your Cmd example I think you leverage the CameraControl.Core library. Is it essential to use it with the .Devices library to be able to achieve this?

It would be really helpful for me to have a demo of how to take a picture or a movie to a folder on the PC. 

Great work,

Hayden

dukus, 2017-10-09:
The demo it is implement the "Saved-to-Disk" option
https://github.com/dukus/digiCamControl/blob/master/CameraControl.Devices.Example/Form1.cs#L77
The file saving is based on camera event "PhotoCaptured" (this event is trigger after photo was captured by the camera no mater if the capture was initialized from software or from camera capture button )
https://github.com/dukus/digiCamControl/blob/master/CameraControl.Devices.Example/Form1.cs#L27

YanYas, 2017-10-10:
Thanks Duka!
How can I choose whether to capture to camera or SD card or both, as per EDSDK?

YanYas, 2017-10-13:
Okay I'm stil struggling with the SDCard, but the trigger and the saving are working.

Thanks again

closed 2017-10-13 by YanYas

dukus, 2017-10-14:
You can set to capture in camera ram with 
`DeviceManager.SelectedCameraDevice.CaptureInSdRam = true;`
in this case no image will recorded in sdcard(even no card is required)

YanYas, 2017-10-17:
I'll need to test it working with correctly triggering capture as we're
looking at an issue with our eventsargs, but the cameras do appear to
respond to it.

Thanks once again Dukus

YanYas, 2017-10-17:
for the record:

[image: Inline images 1]