dukus/digiCamControl #284: CameraControl.Devices (Nuget) Live View Image stuck at 100px X 100px

YanYas, 2017-10-13:
Our attempts to read the Live view data returned only the the above dimensions.

dukus, 2017-10-14:
Which camera model do you use ?
Can you post the used code ?

YanYas, 2017-10-17:
I'm testing with a Canon 550D and a Canon 600D.

So about that code, well... its not all that simple. I'll write a simplifed
example but my code looks like this:

[image: Inline images 1]

Which in code looks the included file.

System.Func<Devices.DigiCamControl.DeviceManager.ForEach__Keep_3, CameraControl.Devices.Classes.PhotoCapturedEventArgs, System.Tuple<Devices.DigiCamControl.DeviceManager.ForEach__Keep_3, string, bool>> Update103 = delegate (Devices.DigiCamControl.DeviceManager.ForEach__Keep_3 Arg 1_3, CameraControl.Devices.Classes.PhotoCapturedEventArgs Arg 2_3) {                tracer.Trace<PhotoCapturedEventArgs, string, bool, bool>(true, 53830, "Update", 53833, ref Arg 2_3, 53852, ref Devices.DigiCamControl.DeviceManager.__const___6, 53861, ref Devices.DigiCamControl.DeviceManager.__const___7, 53862, ref Devices.DigiCamControl.DeviceManager.__const___8);                tracer.Trace(true, 53830, null);                tracer.Trace<PhotoCapturedEventArgs>(true, 53839, "FileName", 53840, ref Arg 2_3);                string File Name104 = Arg 2_3.FileName;                tracer.Trace<PhotoCapturedEventArgs, string>(false, 53839, "FileName", 53841, ref Arg 2_3, 53842, ref File Name104);                tracer.Trace<string>(true, 53868, "ToPath", 53869, ref File Name104);                VL.Lib.IO.Path Result105 = VL.Lib.IO.PathExtension.ToPath(File Name104);                tracer.Trace<Path>(false, 53868, "ToPath", 53870, ref VL.Lib.IO.PathResult105);                tracer.Trace<Path>(true, 53863, "Filename (Split)", 53864, ref Result105);                VL.Lib.IO.Path Directory106;                string Filename107;                string Extension108;                Result105.Filename(out Directory106, out stringFilename107, out Extension108);                tracer.Trace<Path, string, string>(false, 53863, "Filename (Split)", 53865, ref Directory106, 53866, ref Filename107, 53867, ref Extension108);                string Directory109 = Devices.DigiCamControl.DeviceManager.__const___6;                tracer.Trace<string, string, string>(true, 53871, "Filename (Join)", 53872, ref Directory109, 53873, ref Filename107, 53874, ref Extension108);                VL.Lib.IO.Path Result110 = VL.Lib.IO.PathExtension.Filename(Devices.DigiCamControl.DeviceManager.__const___6, Filename107, Extension108);                tracer.Trace<Path>(false, 53871, "Filename (Join)", 53875, ref Result110);                tracer.Trace<Path>(true, 53876, "ToString", 53877, ref Result110);                string Result111 = VL.Lib.IO.Path.op_Implicit(Result110);                tracer.Trace<string>(false, 53876, "ToString", 53878, ref Result111);                tracer.Trace<PhotoCapturedEventArgs>(true, 53843, "Handle", 53844, ref Arg 2_3);                object Handle112 = Arg 2_3.Handle;                tracer.Trace<PhotoCapturedEventArgs, object>(false, 53843, "Handle", 53845, ref Arg 2_3, 53846, ref objectHandle112);                tracer.Trace<PhotoCapturedEventArgs>(true, 53853, "CameraDevice", 53854, ref Arg 2_3);                CameraControl.Devices.ICameraDevice Camera Device113 = Arg 2_3.CameraDevice;                tracer.Trace<PhotoCapturedEventArgs, ICameraDevice>(false, 53853, "CameraDevice", 53855, ref Arg 2_3, 53856, ref Camera Device113);                tracer.Trace<ICameraDevice, object, string>(true, 53847, "TransferFile", 53848, ref Camera Device113, 53849, ref Handle112, 53850, ref Result111);                Camera Device113.TransferFile(Handle112, Result111);                tracer.Trace<ICameraDevice>(false, 53847, "TransferFile", 53851, ref Camera Device113);                bool Value114 = Devices.DigiCamControl.DeviceManager.__const___7;                tracer.Trace<ICameraDevice, bool>(true, 53857, "SetIsBusy", 53858, ref Camera Device113, 53860, ref Value114);                Camera Device113.IsBusy = Devices.DigiCamControl.DeviceManager.__const___7;                tracer.Trace<ICameraDevice>(false, 53857, "SetIsBusy", 53859, ref Camera Device113);                tracer.Trace<string, bool>(false, 53830, "Update", 53834, ref Result111, 53835, ref Devices.DigiCamControl.DeviceManager.__const___8);                return System.Tuple.Create<Devices.DigiCamControl.DeviceManager.ForEach__Keep_3, string, bool>(new Devices.DigiCamControl.DeviceManager.ForEach__Keep_3(), Result111, Devices.DigiCamControl.DeviceManager.__const___8);            };

YanYas, 2017-10-17:
I've tracked down the area that you probably want to see. Look at line 50
in this document.

            bool Movie Is Recording14 = Output13.MovieIsRecording;
>
>             CameraControl.Devices.Classes.LiveViewData Result15 =
> VL.Lib.Control.SwitchNodes.Switch<CameraControl.Devices.Classes.LiveViewData>(Output1,
> __GlobalMethods_3___0.CreateDefault_2(), Output13);
>
>             int Live View Image Width16 = Result15.LiveViewImageWidth;
>
>             int Live View Image Height17 = Result15.LiveViewImageHeight;
>
>

 [VL.Lang.ElementAttribute(ElementId = 54585)]    public Devices.DigiCamControl.CCLiveStream Update([VL.Lang.ElementAttribute(ElementId = 54586)] CameraControl.Devices.ICameraDevice ICamera, [VL.Lang.ElementAttribute(ElementId = 54587)] bool Enable, [VL.Lang.ElementAttribute(ElementId = 54589)] bool Record, [VL.Lang.ElementAttribute(ElementId = 54588)] out byte[] Image_Data)    {        VL.Lang.Platforms.CIL.Runtime.PatchTracer tracer0 = VL.Lang.Platforms.CIL.Runtime.PatchTracer.Get(54444);        if (tracer0 == null)        {            bool Output1 = Enable | Record;            bool Is Connected2 = ICamera.IsConnected;            bool Output3 = __GlobalMethods_3___1.AND(Is Connected2, Output1);            bool Up Edge4;            bool Down Edge5;            Control.TogEdge Input6 = this.TogEdge;            Control.TogEdge Output7 = Input6.Update(Output3, out Up Edge4, out Down Edge5);            CameraControl.Devices.ICameraDevice Output8;            if (Up Edge4)a
            {                ICamera.StartLiveView();                Output8 = ICamera;            }            else            {                Output8 = ICamera;            }            CameraControl.Devices.Classes.LiveViewData l9;            CameraControl.Devices.ICameraDevice I10;            if (Output1)            {                CameraControl.Devices.Classes.LiveViewData Result11 = Output8.GetLiveViewImage();                l9 = Result11;                I10 = Output8;            }            else            {                l9 = this.LV;                I10 = Output8;            }            bool Have Focus Data12 = l9.HaveFocusData;            CameraControl.Devices.Classes.LiveViewData Output13;            if (Devices.DigiCamControl.CCLiveStream.__const___0)            {                l9.MovieIsRecording = Devices.DigiCamControl.CCLiveStream.__const___1;                Output13 = l9;            }            else            {                Output13 = l9;            }            bool Movie Is Recording14 = Output13.MovieIsRecording;            CameraControl.Devices.Classes.LiveViewData Result15 = VL.Lib.Control.SwitchNodes.Switch<CameraControl.Devices.Classes.LiveViewData>(Output1, __GlobalMethods_3___0.CreateDefault_2(), Output13);            int Live View Image Width16 = Result15.LiveViewImageWidth;            int Live View Image Height17 = Result15.LiveViewImageHeight;            Image_Data = Result15.ImageData;            CameraControl.Devices.ICameraDevice Output18;            if (Down Edge5)            {                I10.StopLiveView();                Output18 = I10;            }            else            {                Output18 = I10;            }            bool Have Live View19 = Output18.HaveLiveView;            CameraControl.Devices.ICameraDevice Output20;            if (Devices.DigiCamControl.CCLiveStream.__const__)            {                Output18.StartRecordMovie();                Output20 = Output18;            }            else            {                Output20 = Output18;            }            CameraControl.Devices.ICameraDevice Output21;            if (Devices.DigiCamControl.CCLiveStream.__const___0)            {                Output20.StopRecordMovie();                Output21 = Output20;            }            else            {                Output21 = Output20;            }            bool Have Live View22 = Output21.HaveLiveView;            bool Output23 = !Record;            bool Output24 = __GlobalMethods_3___1.AND(Movie Is Recording14, Output23);            bool Up Edge25;            bool Down Edge26;            Control.TogEdge Input27 = this.TogEdge_0;            Control.TogEdge Output28 = Input27.Update(Record, out Up Edge25, out Down Edge26);            bool Output29 = __GlobalMethods_3___1.AND(Have Focus Data12, Record);            bool Output30 = !Movie Is Recording14;            bool Output31 = __GlobalMethods_3___1.AND(Output3, true);            return Output21 != this.__pad_Id_54567 || l9 != this.LV || Output28 != this.TogEdge_0 || Output7 != this.TogEdge ? new Devices.DigiCamControl.CCLiveStream(Output7, Output28, Output21, l9) : this;        }        return this.UpdateDEBUG(tracer0, ICamera, Enable, Record, out Image_Data);    }    public override void Dispose()    {        return;    }

YanYas, 2017-10-18:
I've tracked down the area that you probably want to see. 
This is at line 50 in the document attached, which is a copy of some of my compiled code.
[DecompileImageStreamer.txt](https://github.com/dukus/digiCamControl/files/1394597/DecompileImageStreamer.txt)

```
             CameraControl.Devices.Classes.LiveViewData Result15 =
 VL.Lib.Control.SwitchNodes.Switch<CameraControl.Devices.Classes.LiveViewData>(Output1,
 __GlobalMethods_3___0.CreateDefault_2(), Output13);

             int Live View Image Width16 = Result15.LiveViewImageWidth;

             int Live View Image Height17 = Result15.LiveViewImageHeight;
```

dukus, 2017-10-18:
That image size is incorrect, and isn't relevant 
The real live image size is given by the jpg which is stored as byte array in ImageData

YanYas, 2017-10-19:
Okay, I get the full texture size like this:

[image]

but my actual Live View image should be 1056 x 704.

YanYas, 2017-11-01:
Okay this has the correct frame size. 

[image]
 Case closed. Thanks

closed 2017-11-01 by YanYas