dukus/digiCamControl #285: How is live view and threading supposed to work?

tebjan, 2017-10-19:
are there any hints on how to best call GetLiveViewImage in a background thread? 
the goal is to put this slow and blocking operation together with image decompression in the background while the UI should be unaffected.

while checking the code i found this:
https://github.com/dukus/digiCamControl/blob/ee1f10a4d326cd73fcc80b0b1d5d973d44837e0e/CameraControl.Devices/Canon/CanonBase.cs#L390

which is probably causing the crashes that we experience when calling it repeatedly from another thread since the timer wants to be on the UIContext.

any help would be much appreciated.

dukus, 2017-10-19:
DigiCamControl use the get liveview image in a another tread like that the UI, i never experienced any issue with it, can you post a code example which cause the crash ?

dukus, 2017-10-19:
Now i observe a thing, you  use the CanonBase.cs which isn't used anywhere, that was just a experiment.
How do you managed to use that class ?

tebjan, 2017-10-20:
ah, i was just browsing the repository, i was not aware that this class is not in use. so never mind, might be a problem on our side with threading then. closing this until further inspection of the problem.

closed 2017-10-20 by tebjan