dukus/digiCamControl #292: Nikon D850 Support?

jordanabyss, 2018-02-20:
Hi there, love the software, but was wondering if we were able to get Nikon D850 Live View support>

Cheers,
Jordan

dukus, 2018-02-20:
Hi, the latest beta already support the D850

jordanabyss, 2018-02-20:
Hi dukus, Yea i've been experimenting with that and the really beta is great (a big upgrade!), albeit quite buggy in places:

- Unable to take images after doing a format all SD through the multi-camera interface
- When project loads with more than 15 images, it freezes. I have to rename the folder with the project images so that the images aren't loaded and it doesn't crash
- Just has random freezes in places (I've got a great PC as well!)

In order to use D850 this week, I was hoping to compile the stable version with the addition of the line {"D850", typeof (NikonD500)} found in CameraControl.Devices/CameraDeviceManager.cs... problem is that the stable version from 2015 throws up a bunch of errors :(

dukus, 2018-02-21:
Can you check this version https://s3.amazonaws.com/download.digicamcontrol.com/digiCamControlsetup_2.0.76.1.exe if the freezing persist ?

jordanabyss, 2018-02-22:
Amazing, that did fix the project loading freeze I was talking about - thanks!

One last question, I'm currently working on using two D850's in parallel to capture images at 2Hz using the multi-camera function (which is an amazing feature btw!). The Live View on multi-camera is super useful for checking the overlap between the two images etc, although I'm finding that the timelapse option within digicam takes very unevenly timed images (even at 1Hz option). I was wondering if it was possible to set-up and trigger the Interval Timer setting found on the D850 through the software. i.e. be able to set interval to 2Hz, hit start, wait 'x' amount of time, then hit 'finish'. I'm saving the images onto the camera SD card as well, so no need to transfer images to digicam. Would significantly donate to the digicam project! :p Cheers
http://onlinemanual.nikonimglib.com/d850/en/18_menu_guide_03_26.html