dukus/digiCamControl #293: Port to Linux

nicolas-raoul, 2018-03-12:
It would be great if digiCamControl could be available on Linux :-)

The steps could be:
1. Compile on Linux
2. Check what features work
3. Write a GitHub wiki page explaining how to get it to run on Linux
4. Make a binary available for download, or create a package for Ubuntu

Keep up the great work!

Quotic, 2018-03-19:
"Compile" on Linux, the problem is that digiCamControl is written in C# and I don't think that dukus get all functions in .net core... but I would also favorite a linux solution :)

shlomiv, 2018-03-25:
I would love a Linux port as well!

Oxydation, 2018-03-25:
First of all, digiCamControl is written with .NET Framework and the GUI is provided via WPF. WPF is only available for Windows applications, not for Linux or MacOS. Furthermore, a lot of functionality is based on libraries to access cameras to interact with them. I did not dig much deeper but the current implementation for PTP/MTP in this software is done by portableDeviceLib - which uses [WinAPI](https://github.com/dukus/digiCamControl/blob/master/PortableDeviceLib/WinApi.cs). 

To give you a much deeper insight how portable digiCamControl is at the moment, I ran the portability analyser in VS2017 (target: .NET Core 2.1/.NET Standard 2.0/Mono): 
[ApiPortAnalysis_digiCamControl.zip](https://github.com/dukus/digiCamControl/files/1845109/ApiPortAnalysis_digiCamControl.zip)
[image]
As you can see, some projects are quite easy to use on a platform independent implementation like .NET Core or Mono. But the majority of projects need a lot of work to make them portable - up to 46% has to be rewritten! Despite PortableDeviceLib is green (because its using DllImport which is portable), it uses e.g. kernel32 for PTP/MPT which is Windows-only. Also, a couple of dependencies from NuGet may not be available in .NET Core or Mono. 
A major part which is not available in .NET Core is a direct replacement of WPF (GUI) - despite being Avalonia somewhat similar to XAML. The GUI has to be rewritten completely if used in another platform (e.g. [GtkSharp](http://www.mono-project.com/docs/gui/gtksharp/) for Mono, [Avalonia ](https://github.com/AvaloniaUI/Avalonia) for .NET Core or via [Web-Interface](https://github.com/ElectronNET/Electron.NET)). 
Check out the zip file (with Excel/Single HTML) for full details which parts/dependencies are not portable.

To make such an application run on another platform, the most important step is to find a way to control cameras platform independent. I am currently working on a project to solve exactly this [problem](https://gitlab.com/define-private-public/SharpCamera/issues/10). If this is solved, major effort can be put into writing a GUI, rewriting functionality from the current digiCamControl. There will be sure some features missing.

scrumit, 2018-04-08:
I haven't looked into this a lot, but [Monocross](https://github.com/MonoCross/MonoCross/tree/master/WPF) has some support for WPF. 

I have to ask; Why linux? Are you carrying a linux laptop around with you?

Might it be better to port to [Xamarin](https://blog.xamarin.com/porting-existing-net-apps-to-android/)? I had been pondering a mobile configuration controller, which is what brought me to this site.

Oxydation, 2018-04-08:
@scrumit I think a lot of people want to use this or similar software on Linux because 
- They don't have a Windows system available
- They don't want to use Windows
- For some scenarios a embedded system (like a RPi, Beaglebone, another IoT board, ...) could be used instead of a Windows machine 

In my opinion I think we have to be prepared for the upcoming future in which we can use software on all major platforms. 

Monocross has, according to their Github [readme](https://github.com/MonoCross/MonoCross), support for iOS, Android, Windows (& Windows Phone/Surface) and more, but it does not mention Linux.

To use [Xamarin ](https://www.xamarin.com/) we would still have to port things to [.NET Standard](https://docs.microsoft.com/en-us/dotnet/standard/net-standard#net-implementation-support). I also thought that Xamarin is more for mobile platforms (Android, iOS and Mac) but not Linux (and Windows). So this would be something interesting to consider if digiCamControl should be accessible via smart phone.

nicolas-raoul, 2018-04-09:
@scrumit I only have a Linux machine at home, and I am not really willing to buy Windows just for one program. I often take pictures at home, and I am hoping that digiCamControl can take pictures at several focus distances automatically so that I can later do [focus stacking](http://digicamcontrol.com/blog/focusstacking) with another tool like enfuse.

I maintain a program (called "CmisSync") using Xamarin (Monodevelop), it is a GUI desktop program and Xamarin can make it run on Linux and Mac (using the graphic libraries of each platform, so there is a lot of platform-specific code). Recently Xamarin means mobile to most people, but it can still be used for desktop GUI programs.

jeffpkamp, 2018-08-26:
What I would like is just a port of the command line tools.   I use WSL because astrometry.net works  much better with that, and I can usually use python to switch between the two, but it would be nice to have all the software I used for my astrophotography on the same operating system, and might be nice to be able to run this on a raspberry pi or something :).

jeffpkamp, 2018-08-26:
and you can ignore that last comment because apparently WSL can execute .exe files :)  Fantastic !

MineHarry01, 2018-10-02:
Love this Tool. My Idea is running a small IoT device with this with a small WebUI to remote-control the Camera.Getting the basic control working at linux would be a nice thing.

thearchitect, 2018-10-19:
As an alternative on Linux you can use [libgphoto2](https://github.com/gphoto/libgphoto2)