dukus/digiCamControl #296: EOS 1000D Live View doesn't work in v2.0.76.1 but does in 2.0.0

NHellFire, 2018-05-05:
Live View comes up as unsupported in v2.0.76.1, v2.0.0 is working fine. Log attached
[digicamcontrol.txt](https://github.com/dukus/digiCamControl/files/1976520/digicamcontrol.txt)

dukus, 2018-05-28:
The Canon cameras are controlled using Canon official SDK, to support ne cameras i need to update this library too. The latest version of the SDK report "Not supported" for EOS 1000D liveview

paulvi, 2018-06-04:
In other words, newer Canon SDK stopped supporting EOS 1000D for Live View.

So this is not issue, of this project.
As solution would be to provide alternative driver/SDK. 
But usually for users it is recommended to have drivers, that are known to work
and substitute driver DLL for older. (e.g. move DLL from 2.0.0 to latest 2.x)

BStrauss3, 2018-06-04:
Likely Canon had difficulties with a ten year old camera and instead of mucking up their software to support it's idiosyncrasies, they just marked it not supported.

With a little work, you can probably cobble together a custom fork of digiCamControl which uses the old Canon library, but it's never going to become mainstream. What I would do is drop in the library, try the compile and keep commenting things out until it compiles. Test it and if it's good enough for your camera and use cases,, you're done. Otherwise you will need to look at what you commented out and code work-arounds.

paulvi, 2018-06-04:
@BStrauss3 Do you mean just replacing SDK DLL (as I suggested to @NHellFire ) would not be enough?
that is with newer SDK DLL updated here, the C# code here also changes, and drop-in of older DLL is not supported?

BStrauss3, 2018-06-05:
Yes.
Later versions of libraries often add new functions which get defined in the code. Older libraries don't have them and throw a linker error so the definition needs to be removed. If the new function is being used, it's more complex.

On Jun 3, 2018, Paul Verest <notifications@github.com> wrote:
>@BStrauss3 Do you mean just replacing SDK DLL (as I suggested to
>@NHellFire ) would not be enough?
>that is with newer SDK DLL updated here, the C# code here also changes,
>and drop-in of older DLL is not supported?

closed 2018-09-23 by dukus