dukus/digiCamControl #297: How to use with OBS Studio

MaxLaLanterne, 2018-05-18:
Hello! 

I need help, DigiCamControl does not work with OBS, I looked everywhere, but nothing works...
 There is a solution?

Thank you!

(Sorry for my english, google trad...)

CCreativeCND, 2018-11-25:
There is a download on the website [here](http://digicamcontrol.com/phpbb/viewtopic.php?f=12&t=898), but it only works for a 32 bit OBS, not a 64 bit OBS.

ComputerCarl, 2019-08-03:
I am attempting to do this as well. I use the DCC webserver and then set up a media service source in OBS. The problem is the output from the webserver is only 640x424. Can this setting be modified somewhere, or is it a limitation of how LiveView works? I.E., if the plugin mentioned above worked with OBS Studio, would it be a full-resolution (or at least HD) image?