dukus/digiCamControl #302: Extract default session parameters to config file

shlomiv, 2018-08-27:
I am trying to set a default "FileNameTemplates" to include the "CaptureName" and could not find where it was defined.

Then I did: https://github.com/dukus/digiCamControl/blob/fcb86271768df066d956e1146f8e6e3aa27358c9/CameraControl.Core/Classes/PhotoSession.cs#L474

Default session parameters should probably be factored out to some json and be configurable.

dukus, 2018-08-28:
The problem not come from that line ...
This is the problem, 
https://github.com/dukus/digiCamControl/blob/master/CameraControl/ViewModel/MainMenuViewModel.cs#L319 the application search for default.xml not for default.json 

I will try to fix for next beta

shlomiv, 2018-11-09:
Any news on this one?