dukus/digiCamControl #305: Deadlocks in async methods

RadoslavGatev, 2018-09-11:
Hi, 
I'm trying to expose an ASP.NET Web API which has to capture and return a photo as an attachment.
But I'm getting a very inconsistent result. Sometimes the first request manages to return the photo but from there on it stops right before composing the final HttpResponseMessage. I believe it's a deadlock caused by the async operations.

It works when using AddFakeCamera.
What I'm doing wrong? Do you know a better way to do this?

```csharp
using System;
using System.IO;
using System.Linq;
using System.Net;
using System.Net.Http;
using System.Net.Http.Headers;
using System.Threading;
using System.Threading.Tasks;
using System.Web.Http;
using CameraControl.Devices;
using CameraControl.Devices.Classes;

namespace WebApplication3.Controllers
{
    public class ValuesController : ApiController
    {
        public string FolderForPhotos = Path.Combine(Environment.GetFolderPath(Environment.SpecialFolder.MyPictures), "Test");

        private TaskCompletionSource<string> TaskCompletionSource = null;
        private CameraDeviceManager DeviceManager = null;

        // GET api/values
        public async Task<HttpResponseMessage> Get(CancellationToken token)
        {
            TaskCompletionSource = new TaskCompletionSource<string>();

            DeviceManager = new CameraDeviceManager();
            DeviceManager.StartInNewThread = false;
            DeviceManager.PhotoCaptured += DeviceManager_PhotoCaptured;

            DeviceManager.ConnectToCamera();
            DeviceManager.SelectedCameraDevice.CapturePhoto();

            token.ThrowIfCancellationRequested();
            var fileName = await TaskCompletionSource.Task.ConfigureAwait(continueOnCapturedContext: false);

            HttpResponseMessage result = new HttpResponseMessage(HttpStatusCode.OK);
            var stream = new FileStream(fileName, FileMode.Open);
            result.Content = new StreamContent(stream);
            result.Content.Headers.ContentDisposition = new ContentDispositionHeaderValue("attachment");
            result.Content.Headers.ContentDisposition.FileName = Path.GetFileName(fileName);
            result.Content.Headers.ContentType = new MediaTypeHeaderValue("image/jpg");
            result.Content.Headers.ContentLength = stream.Length;
            return result;
        }

        private void DeviceManager_PhotoCaptured(object sender, PhotoCapturedEventArgs eventArgs)
        {
            if (eventArgs == null)
            {
                TaskCompletionSource.TrySetException(new Exception("eventArgs is empty"));
                return;
            }
            try
            {
                string fileName = Path.Combine(FolderForPhotos, Path.GetFileName(eventArgs.FileName));
                // if file exist try to generate a new filename to prevent file lost. 
                // This useful when camera is set to record in ram the the all file names are same.
                if (File.Exists(fileName))
                    fileName =
                        StaticHelper.GetUniqueFilename(
                        Path.GetDirectoryName(fileName) + "\\" + Path.GetFileNameWithoutExtension(fileName) + "_", 0,
                        Path.GetExtension(fileName));

                // check the folder of filename, if not found create it
                if (!Directory.Exists(Path.GetDirectoryName(fileName)))
                {
                    Directory.CreateDirectory(Path.GetDirectoryName(fileName));
                }
                eventArgs.CameraDevice.TransferFile(eventArgs.Handle, fileName);
                // the IsBusy may used internally, if file transfer is done should set to false  
                eventArgs.CameraDevice.IsBusy = false;

                TaskCompletionSource.TrySetResult(fileName);
            }
            catch (Exception exception)
            {
                TaskCompletionSource.TrySetException(exception);
                eventArgs.CameraDevice.IsBusy = false;
            }
            finally
            {
                DeviceManager.CloseAll();
            }
        }
    }
}
```

RadoslavGatev, 2018-09-12:
Posted it on StackOverflow as well. I'm preparing an important demo and I need it fixed.
https://stackoverflow.com/questions/52290299/deadlocks-in-async-method-using-taskcompletionsource

dukus, 2018-09-12:
Which camera model do you use for testing ?

dukus, 2018-09-12:
Try this when start capture the photo
```
         DeviceManager.ConnectToCamera();
         while (DeviceManager.SelectedCameraDevice.IsBusy)
         {
                   Thread.Sleep(1);
         }
        DeviceManager.SelectedCameraDevice.CapturePhoto();
```

RadoslavGatev, 2018-09-12:
Nikon D3300

> 
> 
> Which camera model do you use for testing ?

It's Nikon D3300.

RadoslavGatev, 2018-09-12:
> 
> 
> Try this when start capture the photo
> 
> ```
>          DeviceManager.ConnectToCamera();
>          Thread.Sleep(500);
>          while (DeviceManager.SelectedCameraDevice.IsBusy)
>          {
>                    Thread.Sleep(1);
>          }
>         DeviceManager.SelectedCameraDevice.CapturePhoto();
> ```

It definitely works better - I was able to get 2 consecutive photos and then it started working in the old way. I'm certain about one thing - it captures photos and is able to save them into the folder which means that the event handler is fully executed.
The problem is related with either the library itself of my usage of TaskCompletionSource.

RadoslavGatev, 2018-09-14:
I figured it out. Check my answer on StackOverflow: https://stackoverflow.com/questions/52290299/deadlocks-in-async-method-using-taskcompletionsource

closed 2018-09-14 by RadoslavGatev