dukus/digiCamControl #313: "Invalid status" LiveView - CameraControl.Devices.Example

TOAD-Joey, 2018-11-09:
With my Nikon D5100 i get a error message: "Invalid status" on the example project using LiveView. I can capture a photo with the example project but the liveview keeps giving me the error. 

The main project/program liveview works when i build it from source. 

**When i start the liveview in the main project and go back to the "CameraControl.Devices.Example" the liveview works until i restart my camera.** 

Not sure if it's a bug or am i missing something here?

thanossv, 2020-01-14:
Hi Team, 

I am also facing the same issue. I'm posting the details here in case they help.

We have a D3400 Nikon camera with the following firmware:
D3400 firmware:
C:  1.12
LD: 2.017
LF: 1.00

When we tried to enable Live view using the snippet of code at the end of the mail, it threw an exception, with the following details:
exception.ErrorCode = 40964
exception.Message = unknown status

In order to try and see if there is anything wrong with our software, I downloaded the latest standalone application version (2.1.2.0) from http://www.digicamcontrol.com/download and tried to use capture with the same camera. 
When I clicked on the Capture button, the following message appeared at the bottom left, and I couldn’t capture anything at all:
Unknown error. Error code:**2147754003**
Also, the Live View (LB) button was entirely disabled.

The serial number of the camera (as shown inside the camera properties from digicam software) is 0000006369871.

Now, there is also a second camera that works fine with each software. The firmware version is as follows:
D3400 firmware:
C:  1.12
LD: 2.016
LF: 1.00

Using the windows digiCamControl app, it works fine with both capture and Live view.

To be certain that there isn’t anything common cause for this issue, I have already checked/done the following:
•	We tested problematic camera with different cables, usb ports, power supply and laptops, issue persisted
•	We performed factory reset on problematic camera, restarted windows as well but issue persisted
•	I made sure that no other app was using the camera while performing any kind of trial
•	I have added in our code the change introduced in https://github.com/dukus/digiCamControl/pull/339 but the issue is still there

I’m not sure if upgrading to current latest firmware will solve the problem or not, but I’m hesitant to do so as it may break our working camera as well.

If there is any help at all that you could provide to us regarding this matter, it would be greatly appreciated!

Thank in advance and have a good day.

Kind regards,
Thanos S.

```
        private void StartLiveView()
        {
            bool retry;
            do
            {
                retry = false;
                try
                {
                    DeviceManager.SelectedCameraDevice.StartLiveView();
                }
                catch (DeviceException exception)
                {
                    if (exception.ErrorCode == ErrorCodes.MTP_Device_Busy || exception.ErrorCode == ErrorCodes.ERROR_BUSY)
                    {
                        // this may cause infinite loop
                        Thread.Sleep(100);
                        retry = true;
                    }
                    else
                    {
                        MessageBox.Show("Error while starting live occurred (" + exception.ErrorCode + "): " + exception.Message);
                    }
                }

            } while (retry);
            MethodInvoker method = () => _liveViewTimer.Start();
            if (InvokeRequired)
                BeginInvoke(method);
            else
                method.Invoke();

        }
```

TOAD-Joey, 2020-01-14:
@thanossv That is exactly what our problem was. We've updated the camera to the latest firmware with the same result unfortunately. 

We solved the problem by doing the first connection with the camera trough the SDK of Nikon every time a LiveView session is opening. Somehow it connects in a diffrent way of restarts the camera what makes it work. 

`void startNikonSDK()
        {
            Console.WriteLine("Start Nikon SDK");
            try
            {
                NikonManager manager = new NikonManager("Type0005.md3");
                manager.DeviceAdded += manager_DeviceAdded;
                //_waitForDevice.WaitOne();
                manager.Shutdown();
            }
            catch (NikonException ex)
            {
                Console.WriteLine("Nikon SDK error");
                Console.WriteLine(ex.Message);
            }
        }
`

thanossv, 2020-01-14:
@TOAD-Joey thank you for your immediate response!
When searching for the Nikon D3400 SDK I couldn't find any though. I suppose you used the one that exists for D5100, provided by Nikon, right?

TOAD-Joey, 2020-01-14:
That is correct, i used the one from the Nikon website; https://sdk.nikonimaging.com/apply/. I now see that the D3400 is not on the list unfortunately.

thanossv, 2020-01-15:
We got the faulty camera in our hands today (it was in a remote office) and it seems that the change in #339 **does** solve our issue.
The person who tested it did not clear properly the folder with the previous (prior to the fix) version.

slafon11, 2020-11-09:
@TOAD-Joey I've got the same issue with a Nikon D780. I've got the SDK but I don't know what to do with it and where to add your code lines, could you help me with that please ?

TOAD-Joey, 2020-11-09:
@slafon11 Do you have a sample project on Github somewhere where i can add the code or i can contribute? 
I'm happy to help.

slafon11, 2020-11-17:
@TOAD-Joey I'm really new to Github so I don't really know how to deal with it : what should I need to put in the sample project so that you can help me with it ?
Thank you very much !