dukus/digiCamControl #316: Canon 5ds sometimes hangs on EdsDownload call

imagolive, 2019-02-11:
It will take photos for a while and then just sometimes hangs on:

Util.Assert(Edsdk.EdsDownload(directoryItem, size, stream), "Failed to download to stream");

We've tried different cables (including TetherTools), different usb ports (including usb 2.0, 3.0, 3.1). The laptop is fairly new.

DigiCam is run on a separate thread with all camera commands queued to ensure completion before moving on. No errors appear on the camera.

Running out of ideas on what to look at so any suggestions would be appreciated.

kinetixx, 2019-05-25:
Did you figured out what was the problem, i'm having the same problem

dukus, 2019-05-25:
Which version do you use ?

imagolive, 2019-05-25:
No, we haven’t sorted it out and it seems to be happening more often than before.

We’ve tried:

  1.  Tethertools cables and their cableboost to smooth usb power,
  2.  Tried shorter cables,
  3.  Switching off all windows USB power management features for usb ports,
  4.  Put an SD card in the camera (we didn’t use one before),
  5.  Switch off anti-virus in case it was interfering in the transfer of the file,
  6.  Reducing file sizes (jpeg quality) on the camera,
  7.  Tried usb 2 ports (usb 3 seems to have more problems) – this did help a bit,
  8.  Tried both old laptops and new laptops – many clients reported that the camera froze more after upgrading the laptop,
  9.  We tried previous versions of the eos sdk dll,
  10. We upgraded the version of digicamcontrols, but the latest version is a few months old,

It is not clear if anything above actually helps. A client can go for a day without a problem, then have a day where every 2nd picture fails.

It does seem to happen more with high megapixel camera (i.e. 5ds) but it does appear on 24meg cameras also. It mainly seems to be canon, but there have been reports of Nikon’s freezing also (but nowhere near as much).

We cannot reliably repeat the problem in the office, but when it has failed with a debugger, it always points towards the DownloadImage call to the camera – either failing or not returning.

We just recently decided to completely re-write our app, while sticking to “as simple as we can make it”, to see if there is something in our code. Although the code is not too complicated, it does queue the commands to the camera and runs them in a separate background task – waiting for the camera to fully finish one before sending another. It also is a winforms app (moving to wpf) but it worked reliably for at least a year without any major dramas.

kinetixx, 2019-05-25:
I use version 2.1.1.0, USB hub with 4 inputs makes problem (attached log file). Directly to PC 3 cameras works fine
[dcc_log_error.txt](https://github.com/dukus/digiCamControl/files/3219360/dcc_log_error.txt)

imagolive, 2019-05-26:
It sounds like it could be related to power on the usb port. Maybe try a powered usb hub.

kinetixx, 2019-05-26:
the problem is that i use a powered usb hub 3.0 :/