dukus/digiCamControl #332: In SingleCommand, how to start a bulb capture with the set parameters?

shlomiv, 2019-08-04:
Based on http://digicamcontrol.com/doc/userguide/singlecmd#bulb-capture-commands-works-if-the-astro-windows-is-open I can set all the required parameters in the astro window.

However, I don't know how to tell DCC to start the astro capture process (with the desired capture time, iso, number of photos, phd mode and wait etc.)

Is there a way to do such task?

dukus, 2019-08-05:
I will try to recheck in this week the all astro part  and i will give you a working example and/or a new version if there anything to fix..

shlomiv, 2019-08-05:
If you have limited time, I depend on https://github.com/dukus/digiCamControl/issues/333 much more for integrating with programs like SGPro and SharpCap. Thank you!

shlomiv, 2019-08-08:
@dukus It is seems this is already possible when using:

```shell
curl -X GET "http://192.168.1.78:5513/?slc=do&param1=BulbCaptureStart"
````

closed 2019-08-08 by shlomiv