dukus/digiCamControl #335: Single command is missing "Get current image"

shlomiv, 2019-08-08:
We could move between images in a current session using:

```shell
> curl -X GET "http://192.168.1.78:5513/?slc=do&param1=Next_Image"   
> curl -X GET "http://192.168.1.78:5513/?slc=do&param1=Prev_Image"   
```

But there is no way to get the selected image name, i.e.:
```shell
> curl -X GET "http://192.168.1.78:5513/?slc=do&param1=Image_Name"
DSC_0003.nef
```

Maybe even full path with `param2`?

dukus, 2019-08-08:
You can get the whole session information using 
[http://localhost:5513/session.json](http://localhost:5513/session.json)
the last record is the last captured image 
The image can be downloaded using
[http://localhost:5513/image/DSC_0001.jpg](http://localhost:5513/image/DSC_0001.jpg)

dukus, 2019-08-08:
Or you can use more simple
[http://localhost:5513/?slc=get&param1=lastcaptured&param2=](http://localhost:5513/?slc=get&param1=lastcaptured&param2=)

shlomiv, 2019-08-11:
Thank you for the response! Good info on the downloading part, it works well.

Regarding the `lastcaptured`, this does work, with two caveats: 
1. It only retrieves, well, the last captured (I was aiming at retrieving the currently selected imaged.)
2. It only works when the camera is connected to the computer. If it is disconnected it returns a `?` response.

However, that is good stuff for now, thank you!

closed 2019-12-08 by dukus