dukus/digiCamControl #338: Bug setting FNumber on Nikon cameras

samueljfm5, 2019-11-25:
There's a bug in NikonBase.cs (MovieFNumber_ValueChanged and NormalFNumber_ValueChanged) when setting FNumber property.

This property size is 2 bytes, so there should be a cast to UInt16 instead of int.
I tested the current implementation with a Nikon Z6 and it returns error code 0x201B (InvalidDevicePropFormat). 

Changing to UInt16 works well, and property is set successfully.

dukus, 2019-11-26:
Can you check this version https://s3.amazonaws.com/download.digicamcontrol.com/digiCamControlsetup_2.1.1.3.exe

samueljfm5, 2019-12-12:
@dukus thanks for the quick fix. that worked. But should't it be applied to all other camera models, since the aperture property size is only 2 bytes, but you're writing 4 bytes?