dukus/digiCamControl #340: Latest update being detected as a Trojan by Kaspersky

explorer14, 2020-01-05:
Hi,

The latest stable update (v 2.1.2) downloadable from http://digicamcontrol.com/download, is getting flagged as a Trojan by Kaspersky and being removed. Following is the screenshot of the detection:

[image]

The attached file contains the log of the files removed by Kaspersky following the detection:

[Trojan Detected in digiCamControl download.txt](https://github.com/dukus/digiCamControl/files/4023757/Trojan.Detected.in.digiCamControl.download.txt)

Sourceforge.net is [not a safe place](https://www.howtogeek.com/218764/warning-don%E2%80%99t-download-software-from-sourceforge-if-you-can-help-it/) to host the downloadable binaries as they are known to pollute the artifacts with [trojans and malwares](https://www.bleepingcomputer.com/forums/t/700301/downloading-files-from-sourceforge/) of all sorts. 

At the moment, the only safe option for me is to pull the repo and build it from source.

Can you please address this? Amazon S3 or a safe(r) CDN would be the way to go or even just good old FTP on the digicamcontrol.com server. Anything but SourceForge!

explorer14, 2020-01-14:
@dukus any update on the above?

Red-Eyed, 2021-11-30:
@explorer14 
> Sourceforge.net is not a safe place to host the downloadable binaries as they are known to pollute the artifacts with trojans and malwares of all sorts.

So, what's the problem to download it from the github releases page?