dukus/digiCamControl #351: Potential issue with Canon EOS 100D / Rebel SL1

UnforeseenOcean, 2020-09-24:
There is a possibility I am doing something wrong here thus the "potential issue".

Camera: Canon EOS 100D (also known as Rebel SL1)

AF: Tested with it on and off

Issue: Focus stacking not working, focus adjustment not working on Live View

Steps to reproduce the issue:
Connected the camera and turned on the software (which lagged during the start), started Live View and clicked on the controls on the screen to adjust focus.

The focus buttons, while sending the command and making the live view laggy, showed no change despite having pressed the Large Step button multiple times. This happens in both ways, towards the camera and towards the infinity. I can enable autofocus and take photo that way, but locking the focus (on Towards Camera) does nothing. Trying to take focus stacking pictures results in same picture taken multiple times.

hyd76, 2020-09-29:
the focus of the camera need light is it ok with that ?

UnforeseenOcean, 2020-10-01:
I am not sure what you mean by that but the scene is well-lit and the
object is not moving anywhere. It's just the focus adjustment not working
(autofocus works fine)

dukus, 2020-10-01:
Can you check the camera focus drive mode (if there such a option) should not be continuous servo or similar. I can't tell more precisely because i don't own a 100D

UnforeseenOcean, 2020-10-03:
The issue happens with all the modes, AF-OneShot (regular autofocus), AI Focus or AI Servo. It's weird.

UnforeseenOcean, 2020-10-03:
I made it move 512 steps to infinity and it still looks the same, on any modes.

0 steps:
[image]

512 steps (ignore the background):
[image]