dukus/digiCamControl #362: Issue when disconnecting and reconnecting CANON 6D

theweege, 2021-02-10:
Hi,
I'm using digiCamControl to control 3 CANON 6D. If I start the program and turn on one camera, it works perfectly. When I turn on the second camera, it works fine but if I try to shoot a photo with the first one, it says "**Invalid Handle. Failed to get property integer data: propertylid 1281**". When I turn on the third one, it works but the other two don't, and I'm getting the previous error.
If I restart digicam, they work just fine.

caryreno, 2021-06-11:
I am having trouble, getting the same error with a single Canon 6D Mark II, but my error is immediate.

caryreno, 2021-06-11:
I should also add that I have no issues in the distributed digiCamControl application performing this same operation.

On the example application when the camera list populates at load, or if I power the camera on after launching, this is the log output:
New canon camera found !
Pas 1
Pas 2
Error initialize EOS camera object    at Canon.Eos.Framework.Helper.Util.Assert(UInt32 result, String message) in C:\Users\creno\Downloads\digiCamControl-master (1)\digiCamControl-master\Canon.Eos.Framework\Helper\Util.cs:line 28
   at Canon.Eos.Framework.EosCamera.EnsureOpenSession() in C:\Users\creno\Downloads\digiCamControl-master (1)\digiCamControl-master\Canon.Eos.Framework\EosCamera.cs:line 299
   at CameraControl.Devices.Canon.CanonSDKBase.Init(EosCamera camera) in C:\Users\creno\Downloads\digiCamControl-master (1)\digiCamControl-master\CameraControl.Devices\Canon\CanonSDKBase.cs:line 379
Pas 3
Pas 4
Pas 5
===========Camera is connected==============
USB : VID=04a9, PID=32ca
Driver :CanonSDKBase
Name :Canon EOS 6D Mark II
Manufacturer :Canon Inc.
New canon camera found done!
Connection device start
Connection device \\?\swd#wpdbusenum#_??_pcistor#disk&ven_rsper&prod_rts5208lun0&rev_1.00#0000#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}#{6ac27878-a6fa-4155-ba85-f98f491d4f33}
Connection device \\?\usb#vid_04a9&pid_32ca#5&2ca0daf&0&1#{6ac27878-a6fa-4155-ba85-f98f491d4f33}

I get "Invalid Handle. Failed to get property integer data: propertylid 1281" if I attempt to Capture after that.

Thanks.

ITfromHell, 2021-11-07:
Hi, 
because of my missed property ID 1032 we found the solution: 
1. 1032 is 0x408 in hex.
2. This ID is defined at Canon.Eos.Framework/Internal/SDK/EDSDK.cs as 
`public const uint   PropID_FlashCompensation    = 0x00000408;`

But the used Canon EOS R doesn´t have a flash, so this has to bring an error.

@dukus  Solution: Edit CameraControl.Devices/Canon/CanonSDKBase.cs:
[image]
Hint: There are a few more small problems at CanonSDKBase.cs, we corrected meanwhile. I can send you this file.

@caryreno: For your erroring propertyID 1281, 0x501 is defined as
`public const uint   PropID_Evf_Mode                = 0x00000501`
A quick check at Canon EDSDK 13.13.41 says 
0x00000501   kEdsPropID_Evf_Mode   Status of live view
Does you cam support this?