dukus/digiCamControl #366: Nikon Coolpix L830 live view?

TheJaredWilcurt, 2021-03-10:
I've been looking around trying to find a way to get the live view to display on my computer, so I can use it as a webcam for streaming.

## First attempt

1. Plugged Coolpix L830 into DC power (wall)
1. Plugged Camera HDMI out cable into [HDMI Video Capture device](https://i.ibb.co/2MS94gN/IMG-20210104-193708.jpg)
1. Plug Capture device into computer
1. Change Win 10 Privacy setting to allow apps to have camera access
1. Open OBS
1. Add "Video Capture Device" > Select "USB Video" from dropdown
1. It shows the 1 photo stored on internal memory
1. I can interact with the menu and the Up/Down/Left/Right for navigation, but nothing I do takes it out of the Media View section. I can't figure out how to get it to the live video input.

## Second Attempt

1. Install [Nikon Webcam Utility](https://downloadcenter.nikonimglib.com/en/products/548/Webcam_Utility.html)
1. Open OBS
1. Add "Video Capture Device" > Select "Nikon Webcam Utility" from dropdown
1. Connect camera via USB
1. Software does not acknowledge it
1. Updated firmware from 1.1 to 1.1 (I don't know, worth a try). nothing

## Third Attempt

1. Install digiCamControl
1. Connect camera via USB
1. digiCamControl sees camera. I am able to see photos, the serial number, or take a single picture
1. Can't access live view

rhonyabdullah, 2021-08-24:
Hey man @TheJaredWilcurt 
Did you figure out this issue ? I'm facing the same issue and i also have the same hdmi video capture like you, i try to connect and unable to make my camera in live mode.

TheJaredWilcurt, 2021-09-12:
I never heard back from any places online where I posted. So I bought a brio webcam instead. It's okay. If I were to get one now, I'd probably go [Elgato Facecam instead of Brio](https://www.youtube.com/watch?v=AAgHrwsY670).

Might sell the Nikon and invest in a better DSLR in the future after researching which ones are easiest to use as webcams. But for now, Brio is fine.

renansales, 2022-06-12:
Please has anyone managed to solve this problem?

Is there a way to display live image via HDMI or USB cable on COOLPIX Camera?

khatti-bakri, 2022-11-07:
I am also a streamer with the exact same camera and i tried exactly the same things jared did yesterday night.
nut today morning I called up the nikon custumer support and found out their cameras are trash and can't even live view with that capture card. I askked their customer support to buy this camera from me as it is useless but apparently that also does't happen in an "ideal world".