dukus/digiCamControl #382: Using digiCamControl via scripts/command line

SlicerMorph, 2022-05-03:
This tool is awesome, thank you for providing it. 

I am wondering if it is possible to drive the software through command line? Our goal is to automate (to the extend possible) data acquisition for stereophotogrammetry.

dukus, 2022-05-04:
Check this link http://digicamcontrol.com/doc/userguide/remoteutil or http://digicamcontrol.com/doc/userguide/cmd

muratmaga, 2022-05-20:
Thanks, I just started testing this. I can get the command line working with this command:
```
CameraControlCmd.exe /folder c:\temp\capture /filenametemplate test_ /counter 2 /wait 1000 /aperture 16 /capture
digiCamControl command line utility running

New Camera is connected ! Driver :Canon EOS Rebel T6
Canon event 520
Photo transfer begin.
Transfer started :c:\temp\capture\test_0004.JPG
Transfer done :c:\temp\capture\test_0004.JPG
Photo transfer done.
Waiting 1000 milliseconds
```

But when I turn this into a simple text file called test.dccscript and run it via command, I get this error:
```
CameraControlCmd.exe test.dccscript
digiCamControl command line utility running

New Camera is connected ! Driver :Canon EOS Rebel T6
Loading error :Data at the root level is invalid. Line 1, position 1.
```
Can you point at an example of correctly formatted script file?