dukus/digiCamControl #409: Support for Nikon Z8?

struffel, 2023-08-16:
Hello, I saw that the Nikon Z8 is currently not supported by digicamcontrol and when I tried it I got the following error, so it really does not work:

```
PS C:\Program Files (x86)\digiCamControl> .\CameraControlCmd.exe /capture
digiCamControl command line utility running

New Camera is connected ! Driver :Z 8
Error occurred while capturing photo CameraControl.Devices.Classes.DeviceException: Unknown error. Error code:2147754003 ---> System.Runtime.InteropServices.COMException: Ausnahme von HRESULT: 0x80042013
   bei WIA.IDevice.ExecuteCommand(String CommandID)
   bei CameraControl.Devices.Others.WiaCameraDevice.CapturePhoto() in C:\Work\VS\CameraControl\CameraControl.Devices\Others\WiaCameraDevice.cs:Zeile 661.
   --- Ende der internen Ausnahmestapelüberwachung ---
   bei CameraControl.Devices.Classes.ErrorCodes.GetException(COMException exception) in C:\Work\VS\CameraControl\CameraControl.Devices\Classes\ErrorCodes.cs:Zeile 169.
   bei CameraControl.Devices.Others.WiaCameraDevice.CapturePhoto() in C:\Work\VS\CameraControl\CameraControl.Devices\Others\WiaCameraDevice.cs:Zeile 666.
   bei CameraControlCmd.Program.Capture() in C:\Work\VS\CameraControl\CameraControlCmd\Program.cs:Zeile 470.
```

szabadiboti, 2025-03-19:
Check [this](https://github.com/szabadiboti/digiCamControl/tree/master) out.

struffel, 2025-04-21:
This looks promising, thanks you so much!
Though I do need to compile the software myself to get this at the moment, right? Or are there any CI builds or something?

chaoyangnz, 2025-08-02:
> Check [this](https://github.com/szabadiboti/digiCamControl/tree/master) out.

did you test it and help create a pull request to merge to mainline